cedricp/ddt4all #583: Renault abbreviations in DDT4ALL

AndyIreland, 2021-06-23:
hi all, I am stuck on a few of these abbreviations when setting the TCU of an Electric Zoe TCU_GEN2_V3.2 DCM Renault (I think this TCU fitted to some other ICE vehicles in the Renault range too) 

so PROBE - does that stand for something? 

Deactivate Remote Diag - I am presuming when deactivated it stops sending data to Renault or Renault services cannot diagnose the car remotely? 

Deactivate URD - just wondering what URD stands for and what does it do? 

Deactivate PAYD - I presume PAYD stands for 'Pay As You Drive' but what does that actually do?

Deactivate Fleet - what are the implications if this is deactivated?

many thanks if anyone have info on what these mean. 

Andy.

[image]

[image]

KarelSvo, 2021-06-24:
Ask the creator of this ECU file ;-)

``<ddt:Target Name="TCU_GEN2_V3.2" xmlns="http://www-diag.renault.com/2002/ECU" xmlns:ds="http://www-diag.renault.com/2002/screens" email="nicolas.caderon@renault.com" href="TCU_GEN2_V3.2_20151109T114222.xml" upload="11/9/2015" group="TCU"><Description><![CDATA[Status :

AndyIreland, 2021-06-24:
thank you i will give that a go

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif