cedricp/ddt4all #585: Synchronisation ABS wirh dashboard Megane 3 1.9dci 2009

aliloup85, 2021-06-24:
Hi all, and sorry for my poor english;

I've changed my Abs calculator, i had red message error that there is a brake problem. The mechanic did a test at first and the error was gone, but the dashboard displays a new odometer: 350.000km instead of 220.000km!!

I took the car to another mechanic who has corrected the display and it returned 220.000km; but when i do a scan with DDT4all I find a different info inside the ABS calculator: another VIN with more than 350.000km. 

My question: is there any solution on DDT4all to synchronise my ABS calculator with my original dashboard?

Marianpol, 2021-06-25:
There is nothing you can do with the DDT4ALL. You need to download content of a ABS EEPROM memory (it's a 95160), edit mileage stored there and write the data back into the ABS EEPROM. Read and write of a memory can be done using programmer.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif