cedricp/ddt4all #586: ddt4all: crashes on ecu scan

Sig2018, 2021-06-27:
I've switched Hard drive then I reinstalled ddt4.
It works if I go to a specific ECu but If I scan ECUs then ddt4 crashes and shut down.
The curious thing is the error doesn't show in the same percentage of scan, sometimes is sooner than others.
I-m not sure where the crash log is but I have a screen shot of lines and error.
The final line is: AttributeError: 'filter' object has no attribute 'strip'
Please take a look and let me know if you can help me with this issue.
Thank you
[image]

KarelSvo, 2021-06-27:
What exactly from which source did you install?

Sig2018, 2021-06-27:
Hi @KarelSvo , Thank you for your quick answer.
I'm almost sure it is version form drive2.ru forum ddt4all All cars.exe
What I'm not really sure is if I replaced the original ecu file for a new one.
Maybe I did something wrong in new installation but same version was working on previous hard drive.
It has something to do with source or it is ecu or something else?

Sig2018, 2021-06-29:
@KarelSvo  any idea in what may cause that error?
Thank you.

SoaAlex, 2023-05-03:
I have the exact same issue :( Does someone know what it could be ?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif