cedricp/ddt4all #587: Clio 4 Adaptif dönüşe duyarlı sisler - Clio 4 Adaptive rotation sensitive fog activation with ecu tweaker

oktaybilen, 2021-07-01:
Merhaba;
Ecu tweaker ile clio 4 adaptif dönüşe duyarlı sis aktivasyonu yapamıyorum. Sürekli olumsuz cevap alıyorum. Ama diğer özellikleri açabiliyorum. Çözümü nedir?

Hello there;
I can't activate clio 4 adaptive rotation sensitive fog with ecu tweaker. I always get negative answers. But I can open other features. What is the solution?

fparmentier, 2021-07-07:
I believe you must have an updated version of the EMM module.

m0xz0r, 2021-08-03:
Do you mean the air pollution monitor in Rlink? It only works if the [air quality sensor](http://www.catcar.info/renault/?lang=en&l=c3Q9PTUwfHxzdHM9PXsiMTAiOiJNb2RlbCIsIjIwIjoiQ2xpbyBJViIsIjMwIjoiQkhBVSIsIjQwIjoiVXBob2xzdGVyeSAtIEVsZWN0cmljaXR5IiwiNTAiOiI2MCBBaXIgY29uZGl0aW9uaW5nIGFuZCBoZWF0aW5nIGNvbnRyb2wgXC8gQWlyIHZlbnQifXx8bm9wcnM9PTE0OTh8fGJyYW5kPT1SZW5hdWx0fHxub3ByPT0xNDk4fHx0eXBlPT1CSEFVfHxjYXRfaWQ9PUl8fGltZz09fHxncnBfaWQ9PTYwfHxzdWJHcnBfaWQ9PTYyQg%3D%3D) is installed. Otherwise it will always report "good" air quality. Renault has omitted the sensor and disabled the automatic air recirculation on later models after serious complaints about the windows fogging up when bad air was detected during wet weather.

oktaybilen, 2021-08-03:
No, I mean the Fog lights that come on when turning left and right. It is not activated with the Ecu tweaker program.

Android için Outlook<https://aka.ms/AAb9ysg>’u edinin

m0xz0r, 2021-08-03:
Ah, ok, you mean the cornering lights. As with many lighting related options, this requires an EMM with software revision SW14 or higher.

AndyIreland, 2021-08-03:
try doing it with ddt4all instead if you have v14 or higher:
[image]

m0xz0r, 2021-08-03:
The DDT4ALL hack will only work if your car has an EMM SW14 or higher. Do not attempt adding lighting options like cornering lights, "remote lighting" or "follow me home" if your software revision is lower than SW14. I know people whose attempts resulted in a bricked EMM.

AndyIreland, 2021-08-03:
> The DDT4ALL hack will only work if your car has an EMM SW14 or higher. Do not attempt adding lighting options like cornering lights, "remote lighting" or "follow me home" if your software revision is lower than SW14. I know people whose attempts resulted in a bricked EMM.

yep, I have heard or rather read SW14 or under could brick , but only if you change the 'follow me home' parameter and that on SW13 and under - but even so on the side of caution and risk/gamble yep I would be extra careful before I decided to change anything on the lighting for Sw14 and under . 

At the very least I would fire up PyREN and do a backup dump of the lighting EMM ECU just in case it did brick and then try and do a rollback with the save dump to try and rescue it - but there are no guarantees.

oktaybilen, 2021-08-04:
Thank you friens...

closed 2021-08-04 by oktaybilen