cedricp/ddt4all #588: Megane 3 can't get the cruise in bcm

Izimo491, 2021-07-05:
Hello,

A coworker of mine bought a 2012 Megane 3 GrandTour 1.5 dci without cruise control. I've fitted the steeringwheel and button to switch between cruise and limiter. 

When doing the scan it finds 6 ecu's for this car. I can turn de rv/lv on in the tdb cluster window, but the bcm doesn't show the 'cruise' where I can switch it to enable. I've searched in the other bcm's in the top menu and there is one where I can switch cruise, but this can't activate it on this car.

Does it mean it needs a different bcm or is there another way?

Thank you!

fparmentier, 2021-07-07:
You don't need to change something in bcm. Only tdb and injection ecu. I believe it's called RZ047 in Clip/PyRen. I tried it once on a Megane 3 with 1.6 k4m (V40 ecu). But the command wasn't working unfortunately :(

Izimo491, 2021-07-07:
> You don't need to change something in bcm. Only tdb and injection ecu. I believe it's called RZ047 in Clip/PyRen. I tried it once on a Megane 3 with 1.6 k4m (V40 ecu). But the command wasn't working unfortunately :(

Thank you for the reply. I've got it working somehow. I had it already activated in the tdb, but couldn't find it in the other parts where needed to activate, but my coworker said it was working also showing on the cluster.

closed 2021-07-07 by Izimo491