cedricp/ddt4all #589: RLink2 antitheft mode active

Pedrom92ps, 2021-07-05:
Hello, its possible remove antitheft mode? Have Changed unit but antitheft mode activate. Try Change VIN with DDT4ALL but not working.

AlMeu, 2021-07-14:
I can help you, if I get the the unit.

Pedrom92ps, 2021-07-14:
Para onde teria de enviar?

AlMeu, 2021-07-14:
?????

Pedrom92ps, 2021-07-14:
where would it have to ship?



?????

AlMeu, 2021-07-14:
write in clear your email because as you see is nothing.

AlMeu, 2021-08-02:
log in on github and write the email address, do not reply by email, because as you can see it will show your email as *** @ ***. ***

Grazi77, 2023-03-25:
> I can help you, if I get the the unit.

ciao, i have rlink2 blocked with antitheft actived.
help me!!! g.torazza@gmail.com

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif