cedricp/ddt4all #590: code injectors - Megane 3

marcuscl, 2021-07-07:
Hello, i can't find where to code new injectors in DDT4ALL, can anyone help me out? :)

Renault Megane 3 2011
[image]

KarelSvo, 2021-07-10:
What is the Diagversion of your SID305?
Which new codes should be assigned to which cylinder?

marcuscl, 2021-07-10:
> What is the Diagversion of your SID305?
> Which new codes should be assigned to which cylinder?
> The code must be sent divided into 2 parameters.
> And there are 2 different methods, depending on the diagnostic version of the control unit.
> The difference is the emission standard euro 4 or 5. Do you also use Pyren? https://gitlab.com/py_ren/pyren In ddt4all, the code can only be sent via manual input, which is a complex process at euro 5. What language do you speak? With Pyren is easier to write the code and Pyren supports different languages.

Hello, thanks for you reply :)
How do i check the Diagversion? 

I haven't got the injectors yet because, i first got the wrong injectors. I think my car is Euro V. 
I have Pyren and have tried to change to injector codes but it says "There is a scenariumi do not support them!!".

Marianpol, 2021-07-10:
> Hello, thanks for you reply :)
> How do i check the Diagversion?
> 
> I haven't got the injectors yet because, i first got the wrong injectors. I think my car is Euro V.
> I have Pyren and have tried to change to injector codes but it says "There is a scenariumi do not support them!!".

Here is the instruction how to do it in PyRen.
https://www.drive2.ru/l/561334222158037462

marcuscl, 2021-07-10:
> Here is the instruction how to do it in PyRen.
> https://www.drive2.ru/l/561334222158037462

Thanks! :)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

InsanityM, 2023-06-22:
> Here is the instruction how to do it in PyRen. https://www.drive2.ru/l/561334222158037462

Hi, I read that tutorial and the others made by the same guy, but i found a video on youtube and that guy was writing the codes fro injectors on a lagune 2 without the hassle from this tutorial, but for me it is not working. Here is the video https://www.youtube.com/watch?v=G2M7R14Ut4c&t=195s