cedricp/ddt4all #591: Problème connexion Wi-Fi

Cedric317, 2021-07-09:
Bonjour Cédric,

Je n'arrive pas à me connecter sur mon wifi-odb. Le logiciel tente de se connecter sur l'adresse 192.168.0.10:35000 mais il ne se passe rien. Je ne comprends pas car ça fonctionnait avant. Si il y a une solution je suis preneur 😀.

KarelSvo, 2021-07-10:
Which ddt4all version are you using? The current python 3 version is not yet free from all errors.

szarma, 2021-07-17:
Hey @KarelSvo, is there a way to run a previous (presumably python 2) version you refer to? I tried running the app (`python ddt4all.py`) from a cloned repo on macOS and linux and can't connect to my wifi module.  I don't think the module is to blame as it is new and allows me to connect from apps in android (e.g. EOBD Facile).

Cedric317, 2021-07-17:
Hello,
I just tried python 2 and I can connect now. Thank you 🙂

szarma, 2021-07-17:
Hi @Cedric317, thanks for the update. Can you please expand a bit on your comment? Are you on linux/macOS? What python2 version did you run? Can you share the details of the python virtual environment? Cheers

Cedric317, 2021-07-17:
Of course! I use Windows 10 and linux mint. I uninstalled python 3 and installed python 2 on Windows 10. On Linux I have not yet tried but I will take care of it within a week and I will give you a feedback.

3051502, 2021-07-31:
> Of course! I use Windows 10 and linux mint. I uninstalled python 3 and installed python 2 on Windows 10. On Linux I have not yet tried but I will take care of it within a week and I will give you a feedback.

Hi,

Did you just use v1.5 Windows installer?

Thanks

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif