cedricp/ddt4all #593: Rlink2 Upgrade from V3 to V9 and get the issue

Goran-Barznje, 2021-07-13:
Once i upgraded from V3 to V9 i lost many things of my R- link 2 screen, so now i want to downgrade from V9 to V3

Much appreciated for your help image[image][image]

AlMeu, 2021-07-14:
The upgrade from v2.2 or v3.3 to v7.0 is done only at Renault dealers.
Now you are lucky because is not blocked, but a smal distance to block anyway if you try downgrade.
If I have it on my test desk, I could try, but is very risky.

Goran-Barznje, 2021-07-14:
The issue is there is no dealer available in my country to downgrade it

Can you please help to fix that by myself?

Thanks

AlMeu, 2021-07-14:
An email?

Goran-Barznje, 2021-07-14:
> The upgrade from v2.2 or v3.3 to v7.0 is done only at Renault dealers.
> 
> Now you are lucky because is not blocked, but a smal distance to block anyway if you try downgrade.
> 
> If I have it on my test desk, I could try, but is very risky. 
> 
> 

Goran909090@gmail.com

mfarramadan, 2021-08-03:
I want to downgrade from 3.3 to 2.2... is there a way to do so ?

AlMeu, 2021-08-03:
Why you want do that?

Vahidpa, 2022-03-10:
Hi all,unfortunately i have same problem too,but i update it from v3 to v9 directly 
Head unit working now,but many thing don't working, how can i back to v3?
Please help me
@AlMeu

AlMeu, 2022-03-10:
It work luckly, but with many problems.
I think remains you replace the unit.
An email?

Vahidpa, 2022-03-10:
@AlMeu vahidp@gmail.com

sajadkhanizade64, 2022-04-17:
I have same problam.any solution?

Vahidpa, 2022-04-17:
> I have same problam.any solution?

email me please
my email is vahidp@gmail.com

AlMeu, 2022-04-17:
Unfortunately the solution for you is to replace the unit.

sajadkhanizade64, 2022-04-17:
> [sajadkhanizade64](https://github.com/sajadkhanizade64) Unfortunately the solution for you is to replace the unit.

No.i fixed this pb before in 2y ago.but again i upgrade again...:)))

AlMeu, 2022-04-17:
We have an old proverb:
"The pitcher doesn't go to the water often" :-)

sajadkhanizade64, 2022-04-17:
> We have an old proverb: "The pitcher doesn't go to the water often" :-)

Lol:)) i upgrade from v3 to v9.then downgrde to 3 and fix park sensor and camera.but no sound.i had sound on v9:)) i reset ecu and all sounds gone.

AlMeu, 2022-04-17:
It's exactly what I mean, all the time there will be a problem.
Wasting time and nervs for nothing.
Better change it and all problems remains behind.

sajadkhanizade64, 2022-04-18:
@shrlnm hi.i used macro for sound and git this error.can u help me?
[image]

sajadkhanizade64, 2022-04-18:
[image]

sajadkhanizade64, 2022-04-18:
[image]

sajadkhanizade64, 2022-04-18:
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif