cedricp/ddt4all #595: Activating TPMS

mfarramadan, 2021-07-18:
Hello,
I bought 4 TPMS as my car was not equipped with any. Now I installed and activated them in the dashboard and the screen. I aslo noticed that the car did autolearn their ID after a ride. As it is avaikabe in the TPMS screen in ddt4all.
Still i am not getting the tpms readings on the screen. Is there anything else i need to do in ddt4all ?

AndyIreland, 2021-07-18:
have you enabled the TPMS presence in the BCM/UCH ECU on your car? 
[image]

mfarramadan, 2021-07-18:
Yes tried it and it gives me prgm error in the cluster screen.

AndyIreland, 2021-07-18:
Ah right, on the Clio mk4 there were 2 types of TPMS  'indirect' TPMS and 'direct' TPMS  if it was set wrong you got a programme error  like that I think
are they original Renault TPMS valves you have put in the wheels?

mfarramadan, 2021-07-18:
> Ah right, on the Clio mk4 there were 2 types of TPMS 'indirect' TPMS and 'direct' TPMS if it was set wrong you got a programme error like that I think
> are they original Renault TPMS valves you have put in the wheels?

Nope. Copy ones

mfarramadan, 2021-07-18:
But still they are direct ones as they are connected to the air valve.

AndyIreland, 2021-07-19:
cannot think then sorry hope someone else can help

mfarramadan, 2021-07-19:
> cannot think then sorry hope someone else can help

Thanks a lot for ur help.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif