cedricp/ddt4all #596: ecu.py, line 662, in getDisplayValue

andrexweb, 2021-07-19:
The program crashes when I select a parameter with BIN name:

SO: Kali linux 2021.2
Adapter: ELM327 - WiFi( Chinese clone with PIC18F25K80)
ddt4all: latest
Vehicle: Twizy - 2019

```
Traceback (most recent call last):
  File "/home/akane/odbii/ddt4all/ddt4all.py", line 940, in changeScreen
    inited = self.paramview.init(screen, self.screenlogfile)
  File "/home/akane/odbii/ddt4all/parameters.py", line 627, in init
    scr_init = self.initScreen(screen)
  File "/home/akane/odbii/ddt4all/parameters.py", line 988, in initScreen
    self.updateDisplays(True)
  File "/home/akane/odbii/ddt4all/parameters.py", line 1298, in updateDisplays
    self.updateDisplay(request_name, update_inputs)
  File "/home/akane/odbii/ddt4all/parameters.py", line 1341, in updateDisplay
    value = ecu_data.getDisplayValue(elm_response, data_item, request.ecu_file.endianness)
  File "/home/akane/odbii/ddt4all/ecu.py", line 661, in getDisplayValue
    return bytes.fromhex(value).decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
zsh: abort      python3 ddt4all.py
```

`Print Value: ffffffffffffffffffffffffffffffffffffffffffffffffff`

[image]

Note: other displays work fine.
Thank you for your attention and collaboration

KarelSvo, 2021-07-22:
send 21A0
response 61 A0 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF

Hexadezimal FF ist nicht kompatibel mit UTF-8 https://onlineutf8tools.com/convert-hexadecimal-to-utf8
Pyren does not crash, shows an empty line byTwizy BIN. 
[image]

andrexweb, 2021-07-24:
With PyRen it shows the data:

[image]

With ddt4all the following example happens, python 3 vs python 2:

[image]

If an exception handling was added the program would not terminate abruptly. It could be, return the same value or an empty string.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif