cedricp/ddt4all #597: Acc. Wiring

alireza6631, 2021-07-21:
Hi guys
My car is an xfd talisman that has a vizo camera but does not have a radar unit, so I got this unit, but the wiring related to the unit is not available. I want to help, does anyone know the schematic of acc wiring. Special thanks

riu6603, 2021-11-26:
I have exactly same issue with Clio V. If anyone have a knowledge, can you share wiring diagrams and procedure of activation via DDT4ALL ?

Thanks.

AndyIreland, 2021-11-26:
Look on internet for programme called VISU 5.0 (ebay or torrent) for wiring diagrams  - on Clio V OBD2 socket might be firewalled with a gateway that only allows you to view the parameters but not actually make any changes to them and only a renault dealer can change settings - thats the way they seem to be going

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif