cedricp/ddt4all #599: R-LINK 2 NERO

foggiamanuel, 2021-07-27:
Salve, c’è qualcuno che usa risolto con lo schermo nero della kadjar avendo premuto un tasto del display nella parte dello sviluppatore? Grazie infinite

foggiamanuel, 2021-07-27:
Vi prego come avete fatto a risolvere?

AlMeu, 2021-08-01:
Depends if you still have sound and USB ports functioning, I can recover. Otherwise is dead and nobody so far can reflash it.
If is dead remains only to change the unit with one only clean without VIN and antitheft, otherwise will not work.
An email and perhaps I can help you, while I am in Italy now.

foggiamanuel, 2021-08-01:
Si ho tutte e due le porte usb funzionanti. Aiutami. Grazie.

AlMeu, 2021-08-01:
Dammi una mail e ne parliamo in privato.

foggiamanuel, 2021-08-01:
***@***.***

AlMeu, 2021-08-01:
> 
> […](#)

accedi su github e scrivi l'indirizzo email, non rispondere via email, perché come vedi mostrerà la tua email come ***@***.***

foggiamanuel, 2021-08-02:
foggiamanuel@gmail.com

foggiamanuel, 2021-08-02:
Ho scritto l’email. Altro ?

AlMeu, 2021-08-02:
Ho risposto anche alla tua mail ma hai controllato?
Controlla la tua cartella spam.

foggiamanuel, 2021-08-03:
Ho scritto alla tua email dicendo che non sono di Milano. Ho scritto anche Se c’è una procedura specifica per risolvere il problema.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif