cedricp/ddt4all #600: Requête Manuelle ne marche pas - Clio III

monsieur-nick, 2021-07-27:
Bonsoir Cedric,
D’abord, grand félicitations pour ton grand travail pour ce soft, c’est très impressionnants!
J’utilise ddt4all avec Windows 10. J’ai téléchargé la version la plus récente (v2.0 Novembre 2020). La voiture que j’essaye de ‘parler’ avec est une Clio 3 de 2009.
J’utilise un câble ELM327 simple.
Le port COM apparaît bien dans la liste de ddt4all, donc je pense que le driver est bien installer.
L’UCH est reconnu, sauf que la version n’est pas un match parfait.
[image]
Le but – envoyer une requête à l’UCH pour retirer le code ISK à façon de pouvoir programmer un clé vierge.
J’ai essayé d’envoyer une requête pour 21d8 (après avoir changer en mode session 10a0) plusieurs fois, mais ça n’a pas l’air de marcher. Est-ce-que je fais une bêtise ? (Forte possibilité que oui, car je suis un novice à propos de ce type de manipulation, malgré avoir lu beaucoup de poste sur des forums).

J’ai vu sur un vidéo pour une Mégane II que le mode ‘fournisseur’ a était sélectionner avant d’envoyer la requête. J’ai aucune option pour l’UCH de ma voiture- est-ce-que ça devrait y-être ? Faut-il faire une autre manipulation manuelle avant d’envoyer la requête 21d8 ?
J’ai même essayé de modifier un xml (titre UCH_transv_7_0…) avec un peu de copier/coller pour utiliser les fonctions déjà inclus la dedans, mais j’ai pu avoir une réponse de l’ELM.
[image]
[image]
Par contraste, je peux lire les données sur les écrans Infos Clés, Apres Ventes, ou Lecture d’identification du calculateur :
[image]
[image]
[image]
J’ai exécuté le benchmark de ddt4all avec un succès un peu mixte. 86/95 des fonctions ont marché. Le benchmark dit que la version max est 1.4, ce que je penserais aurait était suffisant, mais je vois aussi que certaines fonctions au niveau 1.4 ont échoué. Pense-tu que sa devrait marcher avec ce niveau de compatibilité?
[image]
[image]
Couramment je pense que mon câble ELM est suspect, mais j’en suis pas sûr.
J’ai attaché mes logs les plus récents.
[ecu_ddt.txt](https://github.com/cedricp/ddt4all/files/6887784/ecu_ddt.txt)
[elm_ddt.txt](https://github.com/cedricp/ddt4all/files/6887785/elm_ddt.txt)

Merci en avance pour ton aide à propos de tout ça, je sais que ça fais beaucoup de questions, sans doute un peu idiote.

KarelSvo, 2021-07-28:
The Elmtest results are missing in your elm_ddt file.

monsieur-nick, 2021-07-28:
> 
> 
> The Elmtest results are missing in your elm_ddt file.
> 21D8 is not in the ECU description.
> 

Thank you very much for taking the time the time to reply! My apologies about the log, I think this one should have the test results, somewhere around line 244237 if I'm reading it correctly?
[elm_ddt.txt](https://github.com/cedricp/ddt4all/files/6895253/elm_ddt.txt)

I'm afraid I'm not quite following what you mean about the ecu desciption and screenshot - do you mean that '21d8' is 'nonsense' to this particular ECU? There are a bunch of addresses starting at line 244588 of the log I just attached - are these relevant to the problem?

Is the builder xml appilcation from your screenshot part of ddt4all, or is it a separate tool?

Thanks!

KarelSvo, 2021-07-28:
The answers to ATPPS in lines 244588 to 244599 show that your Elm is OK.  The builder is integrated in the ddt2000 program, does not work standalone. Since the 21D8 command is not documented in your ECU file, I suspect that it is therefore not supported.

monsieur-nick, 2021-07-28:
Ok, thanks for taking a look - that's good news on the cable front at least! I guess I will have to keep hunting for the correct command to get that reserved frame...

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 pas mal !!!

closed 2023-06-02 by Furtif

Gianluck79, 2025-01-30:
buongiorno, mi aggiungo a questa conversazione per chiedere un aiuto. Da  poco ho acquistato una clio 3 phase 2 RSdove il quadro era morto ( presentava solo la spia rossa lampeggiante e all'accensione passava a fisso per qualche secondo e si spegne. facendo la diagnosi con questo software il quadro non dava nessun dato. Ho sostituito con un quadro della serie precedente. Codificato VIN, cambiato KM, settato lingua italiana, modificato da miglia a km. La radio una RENR1B102 mi da sempre LOCKED.
Per semplicità con lo strumento TEXA ho verificato il numero di telaio di tutte le centraline e risulta lo stesso VIN anche per la radio.
Ora con il DDT4DLL la diagnosi all'autoradio non ESISTE, provo anche a selezionare a mano ma nessuno entra. stò usando la versione per MEGANE.
uso interfaccia ELM327 USB.