cedricp/ddt4all #604: Check brake lights warning on dash - 2013 clio MK4

TheCosmicNebula, 2021-08-04:
Hi,

I have been playing around with DDT4ALL with my 2013 Clio, and have had the service light and the warning:

'Check Brake Lights'

Appear on the dash. The brake lights all function correctly, I have upgraded to the PH2 LED units. I never had this issue on my old clio (2014), so am kind of confused. I must have set the values in the EMM_EDISON_SW8 incorrectly, and I'm asking if anyone knows how to solve this, which values need to be corrected.

I would be grateful if anyone knows this, or had access to the default values of the SW8 1.1 Edison unit.

TheCosmicNebula, 2021-08-04:
Gave the EMM ECU a hard reset, seemed to do the trick.

closed 2021-08-04 by TheCosmicNebula

AndyIreland, 2021-08-04:
> Gave the EMM ECU a hard reset, seemed to do the trick.

you were lucky some people have had trouble of bricking ECU when changing lighting settings on SW13 and under - especially if you change the 'follow me home' lighting section

TheCosmicNebula, 2021-08-04:
Yes I have heard. I haven't had issues before today, I thought I'd bricked my ecu, turns out DDT doesn't have a match for the SW10 revision. I saw somewhere that it's recommended to use the SW8 1.1, so I tried that and it was like magic.

Before, when I let DDT auto detect, the EMM module came up red. I tried to change a variable, and it randomised most of the values. Basically made it so that the EMM disabled itself. Then I used the SW8, switched a variable along the lines of lighting_activated, and viola, it started working. Renault wanted £90 just to diagnose the issue. A little bit of searching and tinkering, fixed it for freem

AndyIreland, 2021-08-05:
> Yes I have heard. I haven't had issues before today, I thought I'd bricked my ecu, turns out DDT doesn't have a match for the SW10 revision. I saw somewhere that it's recommended to use the SW8 1.1, [...]

its good when you can fix those things yourself. I thought i bricked my rlink radio on my Zoe as it came up with EOL settings and triangle but with a bit of messing about with settings I got it working again. would hate to think how much renault would have charged me to repair/replace it

TheCosmicNebula, 2021-08-05:
Yeah, i was pulling my hair out for about an hour 😂. Renault want extortionate money to fix stuff like that