cedricp/ddt4all #605: Can't connect to OBD2 in WIFI with ddt4all

manubbk, 2021-08-08:
Hi all

I'm new here, from North of France.
I bought a bought a Grand Scenic 4 Intens from April 2017, with the big screen, and I thought Apple Car Play would be installed but it is not...

So I try to use ddt4all, but I can't see my OBD2 ([this one](https://www.amazon.fr/Foseal-Dispositif-Diagnostic-diagnostiquer-Compatible/dp/B077F8F6TB/ref=asc_df_B077F8F6TB/?tag=googshopfr-21&linkCode=df0&hvadid=454849167565&hvpos=&hvnetw=g&hvrand=1230536409579662278&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9040864&hvtargid=pla-570776036101&psc=1)) in the list.

I can connect to the OBD2 in Wifi from my laptop.
I also managed to use it with the iPhone app EOBD-Facile with no problems.
But from ddt4all it doesn't work. I tried with the firewall and anti-virus disabled but it didn't help.

When I click on "Tests ELM" (down right button), I get this message: 
_Traceback (most recent call last):
  File ".\ddt4all.py", line 1201, in check_elm
    res = elm.elm_checker(port, speed, self.logview, core.QCoreApplication)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 1333, in elm_checker
    print cm[2] + " " + res.strip()
  File "C:\Program Files (x86)\ddt4all\Python27\lib\codecs.py", line 369, in write
    data, consumed = self.encode(object, self.errors)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfc in position 47: ordinal not in range(128)_

Maybe my OBD2 is not compatible?
Any other idea?

Thanks!
Merci !

KarelSvo, 2021-08-08:
give me the file elm.py from your ddt4all program folder

manubbk, 2021-08-08:
Hi

There you are

Thanks!

KarelSvo, 2021-08-10:
exchange and test 

[elm.py.zip](https://github.com/cedricp/ddt4all/files/6962465/elm.py.zip)

manubbk, 2021-08-10:
Thanks but it still doesn't work.

**This time, when I click on "Tests ELM", I get this:**
_v1.0 ATWS [OK]
v1.0 ATWS [OK]
v1.0 ATE1 [OK]
v1.0 ATL1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.0 ATRV [OK]
v1.3 ATS1 [OK]
v1.3 ATS0 [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC0 [OK]
v1.0 ATH0 [OK]
v1.0 ATR1 [OK]
v1.2 ATKW0 [OK]
v1.3 ATD1 [OK]
v1.3 ATD0 [OK]
v1.0 ATAL [OK]
V1.0 ATSP 3 [OK]
V1.0 ATSP 4 [OK]
v1.2 ATIIA 01 [OK]
v1.4 ATSI [OK]
V1.0 ATSP 5 [OK]
v1.4 ATFI [OK]
V1.0 ATSP 6 [OK]
V1.0 ATSP 7 [OK]
V1.0 ATSP 8 [OK]
v1.0 ATPC [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATIB 10 [OK]
v1.0 ATST FF [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SM 1 [OK]
v1.2 ATAT0 [OK]
v1.2 ATAT1 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
v1.4 ATMA [OK]
STN11xx TEST 1 STI [ECHEC]
STN11xx TEST 2 STDI [ECHEC]
STN11xx TEST 3 STPR [ECHEC]
STN11xx TEST 4 STPBRR [ECHEC]
STN11xx TEST 5 STPRS [ECHEC]
Résultat: 47 réussis sur 52

For Media CAN (CAN Line 2) on STN1170, ELS27: Connect OBD pin 3 with 13 and 11 with 12._

**And in the Python console window I get this:**

Tentative d'ouverture du port192.168.0.10:35000 @ 38400
 ATWS                    ATWS

ELM327 v1.5

>
 ATWS                   ATWS

ELM327 v1.5

>
 ATE1                   ATE1
OK

>
 ATL1                   ATL1
OK

>
 ATL0                   ATL0

OK

>
 ATI                    ATI
ELM327 v1.5

>
 AT@1                   AT@1
OBDII to RS232 Interpreter

>
 ATRV                   ATRV
12.8V

>
 ATS1                   ATS1
OK

>
 ATS0                   ATS0
OK

>
 ATCAF0                 ATCAF0
OK

>
 ATCFC0                 ATCFC0
OK

>
 ATH0                   ATH0
OK

>
 ATR1                   ATR1
OK

>
 ATKW0                  ATKW0
OK

>
 ATD1                   ATD1
OK

>
 ATD0                   ATD0
OK

>
 ATAL                   ATAL
OK

>
 ATSP 3                 ATSP 3
OK

>
 ATSP 4                 ATSP 4
OK

>
 ATIIA 01               ATIIA 01
OK

>
 ATSI                   ATSI
BUS INIT: ...ERROR

>
 ATSP 5                 ATSP 5
OK

>
 ATFI                   ATFI
BUS INIT: ERROR

>
 ATSP 6                 ATSP 6
OK

>
 ATSP 7                 ATSP 7
OK

>
 ATSP 8                 ATSP 8
OK

>
 ATPC                   ATPC
OK

>
 ATSH 012               ATSH 012
OK

>
 ATSW 96                ATSW 96
OK

>
 ATSW 34                ATSW 34
OK

>
 ATWM 817AF13E          ATWM 817AF13E
OK

>
 ATIB 10                ATIB 10
OK

>
 ATST FF                ATST FF
OK

>
 ATCRA 012              ATCRA 012
OK

>
 ATCRA 01234567         ATCRA 01234567
OK

>
 ATFC SH 012            ATFC SH 012
OK

>
 ATFC SH 00112233       ATFC SH 00112233
OK

>
 ATFC SD 300000         ATFC SD 300000
OK

>
 ATFC SM 1              ATFC SM 1
OK

>
ATAT0                   ATAT0
OK

>
ATAT1                   ATAT1
OK

>
 ATCM 123               ATCM 123
OK

>
 ATCM 12345678          ATCM 12345678
OK

>
 ATCF 123               ATCF 123
OK

>
 ATCF 12345678          ATCF 12345678
OK

>
 ATMA                   ATMA
B7 00 00 00 00 00 00 00 <DATA ERROR
CAN ERROR
B7 00 00 00 00 00 00 00 <DATA ERROR
CAN ERROR
B7 00 00 00 00 00 00 00 <DATA ERROR
B7 00 00 00 00 00 00 00 <DATA ERROR
B7 00 00 00 00 00 00 00 <DATA ERROR
CAN ERROR
B7 00 00 00 00 00 00 00 <DATA ERROR
B7 00 00 00 00 00 0
BUFFER FULL

>
 STI               STI
?

>
 STDI              STDI
?

>
 STPR              STPR
?

>
 STPBRR            STPBRR
?

>
 STPRS             STPRS
?

>

Does that help?
Thanks again!

KarelSvo, 2021-08-10:
exchange and test

[elm.py.zip](https://github.com/cedricp/ddt4all/files/6963846/elm.py.zip)

manubbk, 2021-08-11:
Still no good.

Here are the error messages when I click on Tests ELM :

v1.0 ATWS [OK]
v1.0 ATWS [OK]
v1.0 ATE1 [OK]
v1.0 ATL1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.1 ATPPS [OK]
v1.0 ATRV [OK]
v1.3 ATS1 [OK]
v1.3 ATS0 [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC0 [OK]
v1.0 ATH0 [OK]
v1.0 ATR1 [OK]
v1.2 ATKW0 [OK]
v1.3 ATD1 [OK]
v1.3 ATD0 [OK]
v1.0 ATAL [OK]
V1.0 ATSP 3 [OK]
V1.0 ATSP 4 [OK]
v1.2 ATIIA 01 [OK]
v1.4 ATSI [OK]
V1.0 ATSP 5 [OK]
v1.4 ATFI [OK]
V1.0 ATSP 6 [OK]
V1.0 ATSP 7 [OK]
V1.0 ATSP 8 [OK]
v1.0 ATPC [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATIB 10 [OK]
v1.0 ATST FF [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SM 1 [OK]
v1.2 ATAT0 [OK]
v1.2 ATAT1 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
v1.4 ATMA [OK]
STN11xx TEST 1 STI [ECHEC]
STN11xx TEST 2 STDI [ECHEC]
STN11xx TEST 3 STPR [ECHEC]
STN11xx TEST 4 STPBRR [ECHEC]
STN11xx TEST 5 STPRS [ECHEC]
Résultat: 48 réussis sur 53

For Media CAN (CAN Line 2) on STN1170, ELS27: Connect OBD pin 3 with 13 and 11 with 12.

And in the Python console :

Tentative d'ouverture du port192.168.0.10:35000 @ 38400
 ATWS                    ATWS

ELM327 v1.5

>
 ATWS                   ATWS

ELM327 v1.5

>
 ATE1                   ATE1
OK

>
 ATL1                   ATL1
OK

>
 ATL0                   ATL0

OK

>
 ATI                    ATI
ELM327 v1.5

>
 AT@1                   AT@1
OBDII to RS232 Interpreter

>
 ATPPS                  ATPPS
00:FF F  01:FF F  02:FF F  03:32 F
04:01 F  05:FF F  06:F1 F  07:09 F
08:FF F  09:00 F  0A:0A F  0B:FF F
0C:68 F  0D:0D F  0E:9A F  0F:FF F
10:0D F  11:00 F  12:FF F  13:32 F
14:FF F  15:0A F  16:FF F  17:92 F
18:00 F  19:28 F  1A:FF F  1B:FF F
1C:FF F  1D:FF F  1E:FF F  1F:FF F
20:FF F  21:FF F  22:FF F  23:FF F
24:00 F  25:00 F  26:00 F  27:FF F
28:FF F  29:FF F  2A:38 F  2B:02 F
2C:E0 F  2D:04 F  2E:80 F  2F:0A F

>
 ATRV                   ATRV
11.2V

>
 ATS1                   ATS1
OK

>
 ATS0                   ATS0
OK

>
 ATCAF0                 ATCAF0
OK

>
 ATCFC0                 ATCFC0
OK

>
 ATH0                   ATH0
OK

>
 ATR1                   ATR1
OK

>
 ATKW0                  ATKW0
OK

>
 ATD1                   ATD1
OK

>
 ATD0                   ATD0
OK

>
 ATAL                   ATAL
OK

>
 ATSP 3                 ATSP 3
OK

>
 ATSP 4                 ATSP 4
OK

>
 ATIIA 01               ATIIA 01
OK

>
 ATSI                   ATSI
BUS INIT: ...ERROR

>
 ATSP 5                 ATSP 5
OK

>
 ATFI                   ATFI
BUS INIT: ERROR

>
 ATSP 6                 ATSP 6
OK

>
 ATSP 7                 ATSP 7
OK

>
 ATSP 8                 ATSP 8
OK

>
 ATPC                   ATPC
OK

>
 ATSH 012               ATSH 012
OK

>
 ATSW 96                ATSW 96
OK

>
 ATSW 34                ATSW 34
OK

>
 ATWM 817AF13E          ATWM 817AF13E
OK

>
 ATIB 10                ATIB 10
OK

>
 ATST FF                ATST FF
OK

>
 ATCRA 012              ATCRA 012
OK

>
 ATCRA 01234567         ATCRA 01234567
OK

>
 ATFC SH 012            ATFC SH 012
OK

>
 ATFC SH 00112233       ATFC SH 00112233
OK

>
 ATFC SD 300000         ATFC SD 300000
OK

>
 ATFC SM 1              ATFC SM 1
OK

>
ATAT0                   ATAT0
OK

>
ATAT1                   ATAT1
OK

>
 ATCM 123               ATCM 123
OK

>
 ATCM 12345678          ATCM 12345678
OK

>
 ATCF 123               ATCF 123
OK

>
 ATCF 12345678          ATCF 12345678
OK

>
 ATMA                   ATMA
F4 83 C8 00 00 00 00 00 <DATA ERROR
F4 83 C8 00 00 00 00 00 <DATA ERROR
F4 83 C8 00 00 00 00 00 <DATA ERROR
CAN ERROR
F4 83 C8 00 00 00 00 00 <DATA ERROR
F4 83 C8 00 00 00 00 00 <DATA ERROR
F4 83 C8 00 00 00 00 00 <DATA ERROR
F4 83 C8 00 00 00 00 00 <DATA ERROR
86 2F C8 00 00 00 00 00 <DATA ERROR
CAN
BUFFER FULL

>
 STI               STI
?

>
 STDI              STDI
?

>
 STPR              STPR
?

>
 STPBRR            STPBRR
?

>
 STPRS             STPRS
?

>

manubbk, 2021-08-14:
Hi guys

What do you suggest?
Should I send back my OBD2 and buy another one?

Thanks

KarelSvo, 2021-08-14:
Looks good. It doesn't get any better with Chinese Elm. If original, then use obdlink sx or ex

manubbk, 2021-09-04:
Hi
I have bought a OBDLink SX and it still doesn't work...
Here is the message I get in the Python console :

Tentative d'ouverture du portCOM3 @ 115200
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
Tentative d'ouverture du portCOM3 @ 38400
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
Tentative d'ouverture du portCOM3 @ 115200
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
Tentative d'ouverture du portCOM3 @ 230400
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
Tentative d'ouverture du portCOM3 @ 57600
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
Tentative d'ouverture du portCOM3 @ 9600
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
Tentative d'ouverture du portCOM3 @ 500000
V├®rfication vitesse de port: 500000
Pas de r├®ponse ELM
Port closed
****************************************
*       Connexion ELM perdue
Port closed

manubbk, 2021-09-04:
Please note I have also tried with the OBDWiz software and it doesn't work either.
I get a "the request is expired" error message.
Thanks for your help

KarelSvo, 2021-09-04:
Use the STN Finder https://www.scantool.net/scantool/downloads/utilities/
if not found now:
USB voltage is missing
Driver not installed properly   https://ftdichip.com/wp-content/uploads/2021/08/CDM212364_Setup.zip
incompatible baud rate set
or hardware defect

manubbk, 2021-09-04:
STN Finder gives me that:

= Retrieving list of available COM ports: =
   COM3

= Attempting to connect at 115200 baud =
  Looking on COM3...nothing

= Attempting to connect at 38400 baud =
  Looking on COM3...nothing

= Attempting to connect at 9600 baud =
  Looking on COM3...nothing

End of search.

manubbk, 2021-09-04:
And I've re-installed the drivers - it doesn't change anything

KarelSvo, 2021-09-05:
Does the updater find your interface? https://www.scantool.net/scantool/downloads/505/obdlink_sx-5.6.19.zip
If not, then there is a hardware defect, exchange it.

manubbk, 2021-09-05:
Here is the message I get in the updater:

00:00:00.005: STN Firmware Updater v1.6.4 started.
00:02:29.206: Connecting to device...
00:02:30.288: Connection error: Connect Failed
00:02:30.288: Firmware upload incomplete. User action required: power-cycle the device and retry firmware upload.
00:02:48.023: Connect Failed

I tried to unplug, wait and retry, but it didn't work any better...

bostu, 2021-11-26:
Hello,
I have the same problem

- 1.st : till this october I had Renault Espace Zen 2016. Back then in 2018 I bought the iLC OBD2 OBD Wireless Car Code Reader from amazon IT and everything works fine. On 6.th of oct this year thiefs stole my car with OBD2 code reader in it.

- 2.nd: now I bought new Renault Espace Initiale 2016. Again I order on amazon UK the same iLC OBD2 OBD Wireless Car Code Reader. When I tried to connect with my PC - it try on default wifi port 192.168.0.10:35000 with all speeds then it put out the message that No ELM327 or OBDLINK-SX detected on COM port 192.168.0.10:35000 ??? I check wifi connection to PC and it is ok, it is connected to OBDIIwifi

- then I tried the newest version of ddt4all - this one doesnt even put out the message that it is no OBD reader, it just wont start

- for check of OBD2 reader I try to connect to it to my phone with OBD HOME mobile app and it connects ok I can monitor RPM, Speed, ...etc

If anyone can help what to do, I would be very grateful.

Thank You

bostu, 2021-11-28:
I checked and try this [uninstaled python 3](https://github.com/cedricp/ddt4all/issues/591) and leave just python 2 from ddt4all and still doesnt connect ... it try to connect with all the speeds and then at the end it says "Connection to ELM lost" "Port closed"
When it try to connect I see on OBD that some kind of connection is established because I see green and blue leds blinking.

poli27, 2022-11-23:
> I checked and try this [uninstaled python 3](https://github.com/cedricp/ddt4all/issues/591) and leave just python 2 from ddt4all and still doesnt connect ... it try to connect with all the speeds [...]

https://cloud.mail.ru/public/LCM2/frc6TSG3Z

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif