cedricp/ddt4all #614: ELS27 connection problem

oolbap, 2021-09-16:
Hi everyone!
I have a problem with my Chineese ELS27. It's a STN1170 chip with FTDI (FT232RL). When I connected by USB to COM1 38400 bps (Windows 10 64 bit) and chose ELM237 it looks ok - I made a test (89/93, v.1.4b). But when I chose ELS27 and 115200 bps (and any other speed) I can't connect - see screen.  I've updated drivers, made a wire inside (pin connection for Renault), I used ELS27 baudrate soft and still nothing.

[image]
g.

AndyIreland, 2021-09-16:
Well for a start it should be a ELM327 with the required PIC chip thats recommended by DDT4all by the developer on github and advise against Chinese clones. 

Also, you only need to turn it into a ELS27 (by re-wiring) if you want to read the multimedia (like an rlink radio) through the OBD2 socket - everything else can be changed without re-wiring and leaving it as a ELM327

oolbap, 2021-09-17:
Thanks for your answer. I explain more detailed:
1. my rewired ELS27, with PIC24HJ128GP504 chip

2. screens with ELM327 test - first at 38400bps, second at 115200bps
[image]

[image]

3. I can't connect to DDT4ALL as ELS27 - see my first post (ddt recognized it as ELM), only as ELM327 is ok
4. I connected as ELM327 to my Grand Scenic II, found ECUs, but any data, can't erase any faults (No DTS)
5. I can't upgrade PIC chip's firmware - now it's 3.3.1, the newest one is 5.6.7

KarelSvo, 2021-09-17:
Which ddt4all version are you using?
show the files elm_ddt and ecu_ddt.

oolbap, 2021-09-18:
Latest github version - Release November 2020 Python 3
Files: [logs.zip](https://github.com/cedricp/ddt4all/files/7189484/logs.zip)

KarelSvo, 2021-09-18:
The python 3 version still contains many bugs. The Elm test looks good in elm_ddt. Test again with the python 2 version.

oolbap, 2021-09-18:
Thx, some success. I downgraded to 2018 version from github. Now I can connect interface as ELS27 115200bps, but ddt can't find any ECU. I tried with card in slot, ignition on and off, Canline 1 and Canline 2. One time ddt found UCH, but wrote somethinig about not full compatibility. And I can't do it again - ddt can't find any ecus now. 
[logs.zip](https://github.com/cedricp/ddt4all/files/7189902/logs.zip)

AndyIreland, 2021-09-18:
why dont you try download DDT4all from here it includes the ECU files and places a shortcut on the desktop to start DDT4ALL and it seems safe, free of malware: - Choose 'All Cars EXE ' 
https://www.drive2.com/b/498093336985338243/

KarelSvo, 2021-09-18:
> Thx, some success. 

Copy the scan results in red and green from the screen log window below.

oolbap, 2021-09-18:
> why dont you try download DDT4all from here it includes the ECU files and places a shortcut on the desktop to start DDT4ALL and it seems safe, free of malware: - Choose 'All Cars EXE '
> https://www.drive2.com/b/498093336985338243/

I don't trust Russian sites. But now I will try ;)

oolbap, 2021-09-18:
> > Thx, some success.
> 
> Copy the scan results in red and green from the screen log window below.
> There are problems when starting the diagnostic session (SDS command) when connecting directly to the UCH.
> few problems when scanning.

Nothing found... I tried with ingnition on/off, Can 1 and Can2.
[image]

AndyIreland, 2021-09-18:
what car are you using it on @oolbap  - its not one of the newer renault's with the gateway firewall protected OBD2 socket is it?

oolbap, 2021-09-19:
@AndyIreland Grand Scenic II from 10/2006

oolbap, 2021-09-20:
Not radio, any other ecu's. I'd like to know something about dtc in my car.

But... I tried Pyren and success. Connected, found 7 ecus and some dtc too. Now I'm trying to learn it.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif