cedricp/ddt4all #615: Enabling reverse sensor screen on R-Link 1

charliehoward, 2021-09-21:
I'm trying to enable this display on R-Link 1, does anyone know where to look?
[image]
I've managed to enable the rear camera screen and there's a space for the sensor display
[image]

AndyIreland, 2021-09-21:
yep I think as i said before try here in the MFD settings : (although you do not seem to have Valeo UPA sensors show up in your list of ECUs on your car so that may be a problem 

[image]

charliehoward, 2021-09-22:
Yeah that enabled the reversing screen but not the sensors, does anyone else know?

AndyIreland, 2021-09-22:
I should imagine the other cars that use the r-link (clio4 , Zoe, later Megane etc) have the parking aid computer in it and then talks to the r-link thats why?

charliehoward, 2021-09-22:
Do they? I got everything required retrofitted, the rear sensors work and you can disable them from the screen. Just don;t know how to enable that display

AndyIreland, 2021-09-22:
Those segments only ever light up on the screen when an obstacle is behind the car ...... but you most probably know that already?

charliehoward, 2021-09-22:
Yes, but I want to enable that part, if you look at my photo you'll see the space is there but there's no car

MellotronScratch, 2021-09-22:
You probably will need another PDC ecu. What's the partnumber of your current PDC ecu?

charliehoward, 2021-09-22:
No don't worry about that it's all there. I just need to find the option to enable it.

AndyIreland, 2021-09-23:
> No don't worry about that it's all there. I just need to find the option to enable it.

if its all there then the UPA ECU should show up when you do a search with the magnifying glass in DDT4ALL - is it?

catalup, 2022-10-16:
@charliehoward @AndyIreland @MellotronScratch 

Hello, I try to find a camera to mount on my Rlink-1. Can you help me with a link or with some information regarding this? The camera is 6v or 12v? Thank you!

AndyIreland, 2022-10-16:
the output from the multi connector on the back of the rlink for the camera is 6v 200ma

[image]

[image]

HK-Zoe, 2023-05-12:
> Got it working. You do have to fit the newer parking module [image]

What settings did you edit to do it?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 not bad!!

closed 2023-06-02 by Furtif

Dorinmomo, 2024-05-16:
[image]
Hello. How did you make the image to work. I have a black screen also.....