cedricp/ddt4all #617: Megane 3 tail lights on with DRL

charliehoward, 2021-09-22:
Does anyone know how to make the rear tail/position lights come on with the DRL's, I remember seeing the option somewhere but can't find it now?

AndyIreland, 2021-09-22:
its in the edison lighting ECU (if its there at all in your ECU fitted to your car)

AndyIreland, 2021-09-22:
example - 
On my car (Renault Clio mk4 2016) its here. - on your megane 3 it could be on another ECU or not even there at all (especially on earlier renaults) 

[image]

srayala, 2022-05-24:
> example - On my car (Renault Clio mk4 2016) its here. - on your megane 3 it could be on another ECU or not even there at all (especially on earlier renaults)
> 
> [image]

To AndyIreland:
Did you scan and have the Edison ECU? I'm asking because I need to select it manually, but unfortunately that specific parameter is in red with NO DATA legend.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif