cedricp/ddt4all #623: Megan 4 multimedia vin

Dark976, 2021-10-29:
Hi. İ am trying to get multimedia vin number from the car by connecting a fat32 usb in the slot and expecting the car to write a file in rlink folder. it does not happen in all my trys. is there any way i can read this value using ddt4all ?

KarelSvo, 2021-10-31:
yes

Dark976, 2021-10-31:
:) very nice of you to gice that much detail

AndyIreland, 2021-10-31:
by ddt4all:

[image]

Dark976, 2021-10-31:
Thank you ver much 👍

[megane 4 radio vin]<https://user-images.githubusercontent.com/49172217/139592997-bc546d62-c7b8-4609-b68c-bbd79d70db23.jpg>

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif