cedricp/ddt4all #624: renault 2 megan 1.5 dci

otpisani, 2021-10-30:
Sorry if I violating policy of use, but before I enter into this world of custom made OBD port, can someone share information if ddt4all and ELM327 USB (hardware unit) actually working on my car model, or where can I find list of supported cars, in PC world would be something like HAL list, Thanks in advance.

AndyIreland, 2021-10-30:
you only need to adapt/wire up custom make if you want to read/change parameters on the radio/navigation/infotainment

otpisani, 2021-10-30:
Can I erase errors with this, presume that I have faulty ECU computer from Delphi, stop signal is flashing after hour or two, and engine is shutting off. When I erase errors with delphi software, I have piece for around 100 km, and like to erase errors by my self, which would be cheaper than investing to fix issue.

KarelSvo, 2021-10-31:
Fault diagnosis with Pyren or Pyclip, repair faults, be satisfied  ;-)

otpisani, 2021-10-31:
Much appreciated, booth of you, case is closed.

closed 2021-10-31 by otpisani