cedricp/ddt4all #626: media nav 4.1.0 connect diag tool :( help pleaseeee.. :(

flame00111, 2021-11-03:
Hi i go in test mode in media nav. And i press clear virgin now its gone... it says in black screen connect diag tool. Please help me.

flame00111, 2021-11-03:
Please help me... thank you

AndyIreland, 2021-11-03:
try copying this LGU file to formatted fat32 usb stick and plug into usb socket of car and wait over half hour with engine running and see if it reloads the firmware for your medianav, it might work?
https://drive.google.com/file/d/1GyD8LgKEVRQcdg0akL4IpCk3zkZGZ2ur/view

flame00111, 2021-11-03:
No bro its update file nothinf happen... :(

AndyIreland, 2021-11-03:
ah right mate this page any help to you?
https://lisandru.eu/automotive/renault-dacia-medianav-tips/medianav-stucked-at-logo-or-inside-of-the-interface-or-pushing-the-clean-virgin.html

flame00111, 2021-11-04:
Yes thank you my friend... its very basic. Just plug in usb and wait a 5 min and tadaaa wince 6.0 :) thank you again my friend have a nice day

nearyux16, 2022-10-26:
@flame00111 how you fix this please help the ling doesnt work anymore

nearyux16, 2022-10-26:
@AndyIreland @flame00111 please help me guys

AndyIreland, 2022-10-26:
> @flame00111 how you fix this please help the ling doesnt work anymore

no, site has closed - here is his facebook profile, maybe send a message through messenger and he may respond? :

https://www.facebook.com/lisandru06

nearyux16, 2022-10-27:
@AndyIreland thanks i send him a message, im waiting for a response. Maybe you know another solution, or know how to fix this?

AndyIreland, 2022-10-27:
if its media nav version 1 (not evolution) 4.1.0 you can try copy the file upgrade.lgu  to a formatted fat32 USB stick and with car engine running and see if this fixes it 

https://drive.google.com/file/d/1GyD8LgKEVRQcdg0akL4IpCk3zkZGZ2ur/view?usp=sharing

flame00111, 2022-10-28:
> @AndyIreland @flame00111 please help me guys

Hey hi bro. This link fix me. You dont need lgu file. You need nk explorer and dmicom test file i give you link. Watch this and your medianav comeback. Note: use usb 2.0 flashdisk dont use 3.0. And wait 3-5 minutes dont be speedy ok. 

https://youtu.be/Be5SnrjW16U

flame00111, 2022-10-28:
> @flame00111 how you fix this please help the ling doesnt work anymore

Hey hi bro. This link fix me. You dont need lgu file. You need nk explorer and dmicom test file i give you link. Watch this and your medianav comeback. Note: use usb 2.0 flashdisk dont use 3.0. And wait 3-5 minutes dont be speedy ok.

https://youtu.be/Be5SnrjW16U

flame00111, 2022-10-28:
Hey hi bro. This link fix me. You dont need lgu file. You need nk explorer and dmicom test file i give you link. Watch this and your medianav comeback. Note: use usb 2.0 flashdisk dont use 3.0. And wait 3-5 minutes dont be speedy ok.

https://youtu.be/Be5SnrjW16U

flame00111, 2022-10-28:
@nearyux16  how you fix this please help the ling doesnt work anymore

Hey hi bro. This link fix me. You dont need lgu file. You need nk explorer and dmicom test file i give you link. Watch this and your medianav comeback. Note: use usb 2.0 flashdisk dont use 3.0. And wait 3-5 minutes dont be speedy ok.

https://youtu.be/Be5SnrjW16U

closed 2022-10-28 by flame00111

nearyux16, 2022-10-29:
Hey guys i try yours solution but anything hapened

nearyux16, 2022-10-29:
@flame00111 i did exatly what he said, but when i put the usb in the medianav, anything hapened

Emanuel2011Paiva, 2022-12-21:
> 

Hmmmm... private video

medianav59, 2023-02-28:
salut jai mon medianav version 4.1.0 bloquer sur connect diag tool comment faire merci

Furtif, 2023-02-28:
> salut jai mon medianav version 4.1.0 bloquer sur connect diag tool comment faire merci

normalement avec ddt4all mais les ELM327 normal ne sont pas ok ou alors il vous faut un avec câble croisé sinon un usb Derlec http://www.derelek.com/nos-produits/

medianav59, 2023-02-28:
J'ai le logiciel ddt4all avec le câble OBD ELM327 
Mais je n'arrive pas j'envoie les vidéos que j'ai fait 
https://youtu.be/FsJfw1GKH3k
 https://youtu.be/ybe0LXX85wQ
https://youtu.be/9sKPYIn6eGI

Furtif, 2023-02-28:
> J'ai le logiciel ddt4all avec le câble OBD ELM327 Mais je n'arrive pas j'envoie les vidéos que j'ai fait https://youtu.be/FsJfw1GKH3k https://youtu.be/ybe0LXX85wQ https://youtu.be/9sKPYIn6eGI

normal que marche pas car les sondes DERLec ou câble croise c'est justement pour accéder a partie can multimédia les ODB Elm327 normaux on accès seulement a une partie can par exemple derlec y a un buton que change can par can2 soit ecus moteur vers ecu multimédia pareil se que fait le russe dans sont tuto https://www.drive2.ru/l/468057702971474426/ soit créer un ODB croisé

Furtif, 2023-02-28:
voir image 
[image]

medianav59, 2023-02-28:
Il faut que je fait quoi pour débloquer du coup

Furtif, 2023-02-28:
> Il faut que je fait quoi pour débloquer du coup

j'ai actualiser le comm image en haut

Furtif, 2023-02-28:
shema 
[image]

medianav59, 2023-02-28:
Wouah trop compliqué pour moi ça 😔 (problème de compréhension) 😒

Furtif, 2023-02-28:
> Wouah trop compliqué pour moi ça 😔 (problème de compréhension) 😒

lol certes c'est pas evident mais vous pouvais toute fois demander de l'aide a Jeremy ici https://www.facebook.com/groups/140606960090892

medianav59, 2023-02-28:
J'avais fait un poste et voici ça réponse 
La procédure est longue pour le remettre en rte et ça ne vaut vraiment pas le coup de le réparer
Achete un mn2 ou 3 a la place tu va économiser de ton temps

medianav59, 2023-02-28:
D'accord merci beaucoup

Furtif, 2023-02-28:
position normal cest 6 - 6 14 - 14 pour croiser sa faut changer 6 - 13 et 14 - 12

Furtif, 2023-02-28:
fautdra fust couper files 6 12 13 14 pas beoin de decouper tous dans la rallonge puis les inverser ou croiser

Furtif, 2023-02-28:
ou y a plus simple voir video ici https://www.youtube.com/watch?v=iDfbmN-OfdM @medianav59

medianav59, 2023-02-28:
Je vais regarder merci

medianav59, 2023-02-28:
Il est possible de faire sur ça 
https://ibb.co/sy17kPP 
https://ibb.co/CBLqB4b

Furtif, 2023-02-28:
> Il est possible de faire sur ça https://ibb.co/sy17kPP https://ibb.co/CBLqB4b

complique par fois les pins sont soudé direct de dans c'est pas évident après la sortie fil noir c'est USB sa na rien a voir avec les pins pour cela que je parlais en rallonge comme sa il n'y a pas a endommager le ODB, voir ceci https://www.amazon.fr/Gorgeri-Diagnostic-Connector-Adaptateur-Connecteur/dp/B082V6RVDQ par exemple mais bon dans mon cas j'ai découper le câble noir car le truc en pastique est monobloc après Jai croise les fils a la place de changer pins comme dans la vidéo plus haut, mais doive avoir des rallonges plus simples a bricoler.

Furtif, 2023-02-28:
voir celui la comme rallonge ou adaptateur https://www.amazon.fr/bbfly-B6-Splitter-extension-Adaptateur-dextension/dp/B073Y519FG peut-etre plus simples a bricoler les pins comme dans la vidéo

Furtif, 2023-02-28:
dans la description Produits fréquemment achetés ensemble y a ton odb je crois

Furtif, 2023-02-28:
apres sa se passe par la dans un ecu de se style 
[image]

Furtif, 2023-02-28:
[image]

si souder comme sa c’est pas trop facile à croiser les pins @medianav59

medianav59, 2023-02-28:
Merci

shrlnm, 2023-03-01:
Hi @Furtif. Do you know which part of configuration (which parameters) should be restored/configured for removing the message "connect diag tool"?  I have a positive experience with R-Link 1 but not with radnav. Anyway, I may generate the autoconfiguration macro for pyren by the vin of this Capture

Furtif, 2023-03-01:
> Hi @Furtif. Do you know which part of configuration (which parameters) should be restored/configured for removing the message "connect diag tool"? I have a positive experience with R-Link 1 but not with radnav. Anyway, I may generate the autoconfiguration macro for pyren by the vin of this Capture

it is basically the micom configuration test the configuration of that ecu saying that if you press the “make virgin” button on micom it deletes the micom configuration files, these files were easy to access with media nav 1 and 2 which were in windows c and now in linux I don't know there's no way to have an explorer or code to enter linux

Furtif, 2023-03-01:
on the ecu screens there is also a make virgin button etc like the res aha and other things that are precisely in micon test .. but if make virgin you don't even have access to the micom test

Furtif, 2023-03-01:
most users who have this problem “connect diag tool” either did shit or pressed “make virgin” or bought a new unconfigured device

shrlnm, 2023-03-01:
@Furtif Thank you 
@medianav59 I may generate a configuration by your vin, but I don't know if someone test it on medianav before.

Furtif, 2023-03-01:
> I may generate a configuration by your vin, but I don't know if someone test it on medianav before.
problem will be how to put the configuration inside the medianav ???? if this is virgin I believe that the configuration can only be recreated via can2 (multimedia can) in WindowsCE versions it could start in inverted memory that is to say having the system in a usb key but it has 4.0.1 I suppose it is WindowsCE, now in new medianav like 3 with linux there complicated

Furtif, 2023-03-01:
look this https://www.youtube.com/watch?v=be8BB6lahvg&list=UUXssGOw0US5JXiktmnN6ImQ
is french but description go be translated

Furtif, 2023-03-01:
that is to say that for a good usb key in inverted memory it would be to have created it before making a mistake with the same data of each MD or partition of windowsCE

Furtif, 2023-03-01:
I don't know if this method of using inverted memory works in MNV 3 maybe copying the device's hdd with dd if=/dev/sda of=/dev/sdb (usb key) but ready I don't know

BenjoKing99, 2026-11-03:
> 12\. if its media nav version 1 (not evolution) 4.1.0 you can try copy the file upgrade.lgu  to a formatted fat32 USB stick and with car engine running and see if this fixes it
>      https://drive.google.com/file/d/1GyD8LgKEVRQcdg0akL4IpCk3zkZGZ2ur/view?usp=sharing

@AndyIreland hey mate, could you please drop a new link if you've still got the synctool patch file? this one is dead. cheers

AndyIreland, 2026-11-03:
> 31\. > 12. if its media nav version 1 (not evolution) 4.1.0 you can try copy the file upgrade.lgu  to a formatted fat32 USB stick and with car engine running and see if this fixes it
>      > https://drive.google.com/file/d/1GyD8LgKEVRQcdg0akL4IpCk3zkZGZ2ur/view?usp=sharing
>      
>      
>      [@AndyIreland](https://github.com/AndyIreland) hey mate, could you please drop a new link if you've still got the synctool patch file? this one is dead. cheers

Hiya , no sorry i havent got them any more but you can try the files on here: 

https://www.gps-rlink.com/medianav/update-software-medianav/