cedricp/ddt4all #627: Who have Icon Clio V with Elm327

safakakdu, 2021-11-03:
Is any one have Clio V with elm327 and ddt4all.
Maybe he can send me Front Wiper and External Light configuration Pages with all parameters.
Example as screenshots

safakakdu, 2021-11-03:
> so can the clio 5 be read / written OK with DDT4ALL then? - I thought the OBD2 socket was blocked by Renault on the later Renault's ?

You can read parameters with ddt4all. 
But for write you need another diagtools I think

closed 2021-11-10 by dawnwhitewater

yuksel967, 2021-11-10:
i don't have clio5 i don't know i haven't had a chance to try

dawnwhitewater ***@***.***>, 10 Kas 2021 Çar, 05:01 tarihinde
şunu yazdı:

> Closed #627 <https://github.com/cedricp/ddt4all/issues/627>.
>
> <https://github.com/cedricp/ddt4all/issues/627#event-5594070876>, or
> unsubscribe
> or Android

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virüs
bulunmuyor. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>