cedricp/ddt4all #628: hi help with ddt4all... :(

flame00111, 2021-11-06:
Hi guys one qiestion. When i start scan witch elm327 bluetooth its find a lot of ecu. And 2 red letters. I see 2 uch/bcm ecus wt...:s why its finding 2 uch/bcm ? (Sorry for my bad english) thank you... one ddt2000 sw 8.3 one is ddt2000 sw 11.0. My car is 2014 clio 1.5 dci 90 hp x98.

[image]

KarelSvo, 2021-11-06:
Why don't you show a photo of the scan result?
Did you select your vehicle model before scanning in order to limit the results to your vehicle type?

AndyIreland, 2021-11-06:
I think red ones are "be extra careful at changing parameters on this ECU" - or 'not a perfect match / what was expected" 

also - if it cannot find ABS ECU automatically , then select it from the list manually you can

flame00111, 2021-11-07:
No my friend. I read xml and json files not perfect match. Im searching what can i do. Thank you

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif