cedricp/ddt4all #631: Renault Talisman DDT4ALL

karodzki, 2021-11-11:
Hello guys. 
I would like to connect to a Renault Talisman using the DDT4ALL. However, when the program is running it shows me "no data" everywhere. When I scan with the ECU, I get this information like on the picture below, does anybody know how to solve it?
[image]

AndyIreland, 2021-11-11:
try increasing 'can timeout' to 600ms see if that helps

karodzki, 2021-11-11:
unfortunately it didn't helped.

AndyIreland, 2021-11-11:
I dont know what version of DDT4all you have but try this one from here with the built in ECU folder and places a shortcut on your desktop and make sure your OBD2 ELM327 has the required PIC chip in it and not a cheap chinese clone.

Choose 'All Cars.exe' 
https://www.drive2.com/b/498093336985338243/

alexunderbases, 2021-11-26:
> try increasing 'can timeout' to 600ms see if that helps

Hello!
Can you please teach me how to reset the welcome settings of the DRL?
A few days ago, i tried to replace the high beam halogen bulb and replace it with LED.
In DDT4ALL, I changed the settings from HALOGEN to LED, but I got the "Check Lights" error. Seeing this, i came back to that setting and changed it back to halogen, restarted the car and the error dissappeared.
BUT, this left me with something strange happening with my DRL Welcome sequence:
When I unlock my car, the DRL lights up very very bright and very fast, after wich, it returns to it's normal brightness when the engine is off and when the car is running.
I can be reached on Whatsup at +40736284508 or on Facebook @ https://www.facebook.com/alexandru.gabriel.31392410
RENAULT TALISMAN 2016 ZEN
Thank you in advance!!!

AndyIreland, 2021-11-26:
Thats a tough one - if you previously done a backup of the UPC-EMM with PyREN programme *before* changing any parameters with DDT4ALL  then possibly you could have used the .txt file to do a rollback in pyRen to put it back to defaults .. did you do a backup/dump with PyREN before hand?

alexunderbases, 2021-11-27:
> Thats a tough one - if you previously done a backup of the UPC-EMM with PyREN programme _before_ changing any parameters with DDT4ALL then possibly you could have used the .txt file to do a rollback in pyRen to put it back to defaults .. did you do a backup/dump with PyREN before hand?

Unfortunately, no... . Is there no other way to return to factory settings? What if I disconnect the battery and leave it for 5 minutes?

KarelSvo, 2021-11-27:
All changes are recorded in the ddt4all log files.  

A Pyren ddt-dump is helpful if you trigger a virgin command without saving the original configuration beforehand.

alexunderbases, 2021-11-27:
@AndyIreland @KarelSvo 
Maybe this can hel
[LOG.TXT](https://github.com/cedricp/ddt4all/files/7611454/LOG.TXT)
p.

AndyIreland, 2021-11-27:
> @AndyIreland @KarelSvo Maybe this can help. [LOG.TXT](https://github.com/cedricp/ddt4all/files/7611454/LOG.TXT)

I dont think so - normally a PyREN does a TXT file that looks like this when it create's a Dump file looking like this:

224B31:62 4B 31 00
224B24:62 4B 24 07 63
224B3C:62 4B 3C 02
224B3E:62 4B 3E 00
224B43:62 4B 43 AA
224B40:62 4B 40 00

and so on

KarelSvo, 2021-11-27:
ddt4all log file original configuration
Lights - Configuration	Request : DataRead.E_E_LowBeamType
		{"E_E_LowBeamType": "dt_ENUMLightType_HALOGEN"}
Lights - Configuration	Request : DataRead.E_B_Cornering_Inhibition
		{"E_B_Cornering_Inhibition": "0"}

luis8097, 2022-03-25:
@AndyIreland why dont you explain how to make backuo with Pyren. On every thread and every problem you are mentioning that people should do backup in pyren before changing anything in ddt4all.

Wouldn't it be better, instead of writing comments that would not help anyone, to write a short tutorial for everyone on HOW TO MAKE BACKUP WITH PYREN before MAKING ANY CHANGES IN DDT4ALL. I bet people wpuld be grateful for your effort to help.

KarelSvo, 2022-03-25:
The dump will be created automatically, if this option is enabled at startup.  https://gitlab.com/py_ren/pyren/-/wikis/Using%20launcher

AndyIreland, 2022-03-25:
> @AndyIreland why dont you explain how to make backuo with Pyren. On every thread and every problem you are mentioning that people should do backup in pyren before changing anything in ddt4all. [...]

Why is the onus on me to write a tutorial on it? - why dont you go search like i did originally on a plethora of sites . i might if I feel like it one day doing a video or blog but other than that there are plenty of other articles on the Internet about PyREN and how to perform dump files. 

I still think its helpful to write "before you make any changes do a backup in PyREN2 rather than not mention at all about backing up. 

how 'grateful' will people be if I write up a blog / video  taking time out to do it ? .. are we talking money-wise / donation 'grateful' are we? :)

luissilva1971, 2022-04-03:
Is there a toturial how to restore ecu config from a previoud saved dump?king regards

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif