cedricp/ddt4all #635: Renault Talisman DDT4all and me as a stupid user.

Schluchtmann, 2021-11-12:
Upgraded my talisman with an new usb reader so i could use android auto. But than i was that stupid that i pressed the video high button in the developer screen. Now my screen is black. I found this guide.
I really don't get how to run it. Every python i have tells me name error.
I have the reader DT4ALL works fine. But i really don't get the steps in this.
Maybe somebody would like to help here.

How to repair R Link 2 with a black screen after HIGH MID or VIRGIN MODE button
Run one of appropriate command macro. You need the python 2.7.x installed and the pyren code unpacked to any folder. For running the macro you have no need to install any database.

run the launcher _pyren_launcher.py
choose COM port of your ELM or its IP address in case of WiFi
press “Macro” button
choose one of the next 4 files
rlink2-HIGH1-screen.cmd
rlink2-HIGH2-screen.cmd
rlink2-MID1-screen.cmd
rlink2-MID2-screen.cmd
wait while the macro is executing. 

This is the link to the guide. It is far down on the page.

https://lisandru.eu/automotive/ddt4all-hacks/r-link-functionalities-activated-by-ddt4all.html
Alle The Best....

Michael from Denmark

KarelSvo, 2021-11-13:
Why don't you show your error message?

Schluchtmann, 2021-11-15:
To be honest.  Probably because I have almost 0 knowledge of python.  name error name is not defined. I probably launch the _pyren_launcher.py wrong. I am starting python and try to run the script.  But than I get the name error. I think I try to follow as good as possible.  So probably I really do something wrong in the basics. It also looks like it is possible to run this Macro from ddt4all. But I can't figure out how. Sorry for the lots of questions.  All the best Michael

Schluchtmann, 2021-11-16:
> Why don't you show your error message?
To be honest. Probably because I have almost 0 knowledge of python. name error name is not defined. I probably launch the _pyren_launcher.py wrong. I am starting python and try to run the script. But than I get the name error. I think I try to follow as good as possible. So probably I really do something wrong in the basics. It also looks like it is possible to run this Macro from ddt4all. But I can't figure out how. Sorry for the lots of questions. All the best Michael

Schluchtmann, 2021-11-18:
Figured it out :-) As i said. Just me that had an issue to understand the order.
Works for me now. :-)

closed 2021-11-18 by Schluchtmann