cedricp/ddt4all #639: [Query] - WIFI Connection iLC OBD2

globsecure, 2021-11-23:
I made the purchase of an "iLC OBD2 OBD Wireless Car Code Reader Scan Tool" but the DDT4ALL software when trying to connect, informs that it is not compatible, I believe that the version I bought does not edit variables. It is possible?

I followed this video: 

https://youtu.be/o6Uue6GMGjY?t=59

bostu, 2021-11-26:
Hello,
I have the same problem

- 1.st : till this october I had Renault Espace Zen 2016. Back then in 2018 I bought the iLC OBD2 OBD Wireless Car Code Reader from amazon IT and everything works fine. On 6.th of oct this year thiefs stole my car with OBD2 code reader in it.

- 2.nd: now I bought new Renault Espace Initiale 2016. Again I order on amazon UK the same iLC OBD2 OBD Wireless Car Code Reader. When I tried to connect with my PC - it try on default wifi port 192.168.0.10:35000 with all speeds then it put out the message that No ELM327 or OBDLINK-SX detected on COM port 192.168.0.10:35000 ??? I check wifi connection to PC and it is ok, it is connected to OBDIIwifi

- then I tried the newest version of ddt4all - this one doesnt even put out the message that it is no OBD reader, it just wont start

- for check of OBD2 reader I try to connect to it to my phone with OBD HOME mobile app and it connects ok I can monitor RPM, Speed, ...etc

If anyone can help what to do, I would be very grateful.

Thank You

bostu, 2021-11-28:
I checked and try this [uninstaled python 3](https://github.com/cedricp/ddt4all/issues/591) and leave just python 2 from ddt4all and still doesnt connect ... it try to connect with all the speeds and then at the end it says "Connection to ELM lost" "Port closed"
When it try to connect I see on OBD that some kind of connection is established because I see green and blue leds blinking.

Sharuman81, 2022-03-24:
Hi, I am having exactly the same issue. I tried with two computers and two different ddt4all versions.
Any idea to help us, please? I would like to recover the screen in my Megane IV Zen, because I broke it touching the developer mode.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif