cedricp/ddt4all #640: No launch DDT4All

Jean9740, 2021-11-23:
Hello, I am contacting you because I have a problem with the DDT4All software. My version of the operating system is Windows 7 32 Bits but installed on a computer containing 64 Bits hardware. When launching the DDT4AAll software, I have the 32/64 Bits version error. thanks for the help

AndyIreland, 2021-11-23:
hello @Jean9740  - try downloading DDT4ALL from here instead (its a russian site but program can be configured to English in setup - choose 'All cars EXE' ) and it should be safe of malware etc.
it will put a handy shortcut on the desktop and includes the all important vehicles and ecu folders and should do 32bit and 64bit windows
https://www.drive2.com/b/498093336985338243/

Jean9740, 2021-11-23:
hello AndyIreland thank's you for your help. But i don't working. I have error : %comspecs%

Jean9740, 2021-11-23:
Hi, i have try your command but : Windows not found set, please try

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/discussions

closed 2023-06-02 by Furtif