cedricp/ddt4all #641: Rest.activate for renault  captur x87

dexter2111, 2021-11-25:
Hello,
I activate start engine remotely(with ddt4all),but i don't have the button to start the engine on the  remote.
I also activate the starting scheduled menu, on audio unit, but the engine does not start. 
In Russia this function is working , but in the Europe no. :(
Please some help.

AndyIreland, 2021-11-25:
try long pressing on each the remote keycard buttons and see if anything happens

dexter2111, 2021-11-26:
Hi AndyIreland,
Thank you for answer, i see all your tutorials about ddt4all, and changing light setting on clio 4. Great job.
I try the long press on the remote key,is not working, with the car out of gear.
The car has activated, "press clutch" to start the engine.

AndyIreland, 2021-11-27:
> Hi AndyIreland, Thank you for answer, i see all your tutorials about ddt4all, and changing light setting on clio 4. Great job. I try the long press on the remote key,is not working, with the car out of gear. The car has activated, "press clutch" to start the engine.

Hi Dexter thanks for that. 
Try this see if it works it will allow the car to start without pressing either the clutch or the brake pedal. Find your particular HFM ECU Module and change the start with pedal press setting to 'FALSE'  - good luck
[image]

Furtif, 2023-02-21:
This is available in micom test is more easy activation for the stuf by micom

fortizen, 2023-05-31:
> Hello, I activate start engine remotely(with ddt4all),but i don't have the button to start the engine on the remote. I also activate the starting scheduled menu, on audio unit, but the engine does not start. In Russia this function is working , but in the Europe no. :( Please some help.

Can you show me how to Activate Remote Engine Start because i tried in DDT4ALL and it had an Error Trying to Write the setting.?

AndyIreland, 2023-05-31:
> Can you show me how to Activate Remote Engine Start because i tried in DDT4ALL and it had an Error Trying to Write the setting.?

what year renault car have you got? - it might be security gateway stopping the changes

fortizen, 2023-05-31:
> what year renault car have you got? - it might be security gateway stopping the changes

I have a 2020 (Phase 1) Megane R.S. Trophy 1.8 300 (Manual)

AndyIreland, 2023-05-31:
> I have a 2020 (Phase 1) Megane R.S. Trophy 1.8 300 (Manual)

ah right then its more than likely got the security gateway - you will have to unlock it with a Token to make changes

AndyIreland, 2023-05-31:
have to go to renault main dealer - or if you prove you are a registered mechanic can purchase token from here :

https://asos.renault.com/#/presentation

fortizen, 2023-06-01:
> have to go to renault main dealer - or if you prove you are a registered mechanic can purchase token from here :
> 
> https://asos.renault.com/#/presentation

[image]

It says that my Megane Doesen't have a Securuty Gateway

Furtif, 2023-06-01:
[image]

💁🏻‍♂️

fortizen, 2023-06-01:
> [image]
> 
> 
> 💁🏻‍♂️
> 

What does this mean?

AndyIreland, 2023-06-01:
er Remote Engine Start!

fortizen, 2023-06-01:
> [image]
> 
> 
> 💁🏻‍♂️
> 

What Software is that and What OBD Tool do you use for that? I Have a Normal ELM 327 Wi-Fi OBD-2 Adapter that i Use with ECUTweaker (DDT4all For Android), Can i Enable RES with it? I Couldn't Enable it with ECUTweaker.

dexter2111, 2023-06-01:
micom test

Furtif, 2023-06-01:
Need only ser it to “on” and parametres of is show in medianav as a new icon of ..

Furtif, 2023-06-01:
[image]

normally this is active in cold countries like Russia etc...

dexter2111, 2023-06-01:
@Furtif 
In know that in Russia, is available.
But in Europe, I'm not sure if I need a module.
The values ​​can only be changed with the engine running.
After activation, the icon appears, the time can be set, but nothing happens.
I asked on the Russian forums, where more complicated changes were made than activating and deactivating some functions.
But I did not receive any answer.
I have been trying to find an answer for two years.

Furtif, 2023-06-01:
> @Furtif
> In know that in Russia, is available.
> But in Europe, I'm not sure if I need a module.
> The values ​​can only be changed with the engine running.
> After activation, the icon appears, the time [...]

I activated this on my renault captur to see, the button appeared but I didn't test it… I think the days and time are configured in the medianv for the car to turn on by itself… I don't know how I didn't test it, I can't say that it works, but I think it does aha it's off if I turn it on it works with the app aha, the only thing that's not ok on the renault captur (x87) and the 4wd because it's not 4x4 and doesn't have modules for that. In theory the factory is the same depending on where you export the configuration differs in standard the captur does not have the option of recoil camera and sensor but if you activate the sensors they are present (behind) and the wires to connect the camera are also present

Furtif, 2023-06-01:
Need a test better

Furtif, 2023-06-01:
on the other hand and since this “res” is used in medianav I assume that the ecu in question is can2 to access this ecu you need crossed odb 2 or special cable, read in this issue https://github.com/cedricp/ddt4all/issues/626#issuecomment-1448101286

Furtif, 2023-06-01:
[image]

basically it is activated by ddt4all need obd compatible can2 this image is part of the ucl xx ecu configuration, or you can go directly through the micom

dexter2111, 2023-06-01:
@Furtif 
I modified my WIFI ELM OBD, I can access both can speeds (low/high) ECU, and MEDIA.
The "RES", and other light functions are activate with ddt4all.
Maybe it only works in cold climate countries.( my remote key does not have an engine start button,she only has: lock/unlock,trunk, light.
I tried in the winter at temperatures below -5 degrees, without success.
I am attaching the pictures with the settings made 2 years ago.
[image]
[image]

Furtif, 2023-06-01:
🤷🏻‍♂️I haven't tested anything because I'm afraid the car will start on its own and use up the gas tank lol.

but I still have to see if it works, I test it and deactivate it again.

AndyIreland, 2023-06-01:
did you see there is another setting in the Body control module / bcm monitor config which i am predicting precond means 'precondition' and another saying for RES remote engine start???
[image]

theres a MMI start setting in the FAI MMI configuration - I am predicting the MMI could stand for multi media interface - meaning the medianav screen???
[image]

then you have AT RES authorization which I presume if you have automatic transmission then you have to set that to 'true too' and posibly set the temp settings at the top:
[image]

Think you also have a setting in the lighting EMM edison ECU for RES feedback which presumably flashes your headlights to tell you that RES has been triggered.

fortizen, 2023-06-02:
Are we Talking about R-Link 2 Here?

AndyIreland, 2023-06-02:
well those screenshots I put up are for the BCM ECU so independent of any medianav or rlink settings. 
But on my car 2013 Renault zoe EV its rlink1
I cannot see any RES settings in that

The wife has a 2016 renault Clio mk4 with a different radio -  a Medianav
That definately has the RES in MICOM Manager and DDT4ALL settings

Furtif, 2023-06-02:
> well those screenshots I put up are for the BCM ECU so independent of any medianav or rlink settings. But on my car 2013 Renault zoe EV its rlink1 I cannot see any RES settings in that
>
> The wife has [...]

the parameters mentioned above may be fine, but if what makes it work is off, it gets complicated.

I can have a series of code:
```
if xxx then {
read and use other values
....
}
else {
skip...
}
```
it will have to be on in micom or in UCLx.xx ecus all that.

Furtif, 2023-06-02:
if calculator get
if res.on {
values
}
not need else because only check if is on or off is ok

Furtif, 2023-06-02:
or simplify ...
if res == off all others ecus is skipped

Furtif, 2023-06-02:
a configuration can be 100% good but it is only taken into account if its use is active, I think that the parameters given above at https://github.com/cedricp/ddt4all/issues/641#issuecomment-1572908648 are data that they are configured just by clicking on medianav or r-link https://github.com/cedricp/ddt4all/issues/641#issuecomment-1572566934 (I don't know if r-link and so I didn't see it up close)

Furtif, 2023-06-02:
I don't believe that renault has done this to having to use an ODB2 for each time it is used ... a Russian or Nordic country will not put an obd2 in each configuration so it is in the medianav car radio with easy access (if enabled ) and with an easy configuration, otherwise the car leaves the factory with pre-programmed dates and times and to change it you have to go to a dealer ???

Let's be consistent...

Furtif, 2023-06-02:
if none of this works I assume it has to do with the MAP #### or dual check configuration

europa uses the F4 europe west and nothing to do with the T4 this will be data that can be exchanged with the graphical interface via medianav... otherwise everyone who would buy a renault in nordic countries would be given a ddt2000 or ddt4all... no I believe the constructor would give it.

Furtif, 2023-06-02:
the dual check would be something more or less like this

```
if res.on and Map.region.Fx (nordic country) then
....
```

Furtif, 2023-06-02:
anyway I will do some tests, to clear the doubts.

Furtif, 2023-06-02:
cant not press Make virgin in micom 
[image]

Furtif, 2023-06-02:
for save changes Res -> off to Res-ON press Hard-Reset it save changes and reboot mediaNav

AndyIreland, 2023-06-02:
> cant not press Make Virgin in micom, this make you mediaNav formated and ask for diag-tool or CAN2 [image]

yeas seconded .. **do not press Make Virgin!!!!**

Furtif, 2023-06-02:
https://github.com/Furtif/MediaNavIGO/tree/master

for get inits to go to micom 
[image]

some do not know how to create an empty ini.

Furtif, 2023-06-02:
[image]
set usb stick and press Gen Micon Ini's

Furtif, 2023-06-02:
There are many issues here that shouldn't even be issues... but that's it, you use the issue mode to ask questions, normally an Issue on github is equal to a problem in given source code, and not exactly an issue, or bad use of it , by the way @cedricp  should open the "discussion" mode this avoids all type of question in issue mode.

@cedricp (sorry for tag)

cedricp, 2023-06-02:
Discussion activated !

Furtif, 2023-06-02:
parfait
regarde je le avais mis la https://github.com/Furtif/ddt4all/discussions/8
... mais c'est ton repos pas le mien

cedricp, 2023-06-02:
En effet, le probleme est que j'ai pas le temps de m'en occuper, tu veux un acces a ce repos ? Ca serait cool que quelqu'un fasse un peu de menage :)

Furtif, 2023-06-02:
autre chose tan que je te tiens @cedricp change ceci https://github.com/cedricp/ddt4all/blob/master/.github/FUNDING.yml#L3
par ton lien paypal

cedricp, 2023-06-02:
Ok, quand j'aurais 2 mn ;)

Furtif, 2023-06-02:
et change furtit par le tien aussi desolé a force de faire choses etranges lol ... il y a que echappe mais ca va personne ma donner un centime pour cela

cedricp, 2023-06-02:
Oui, il y est chez moi

Furtif, 2023-06-02:
chelou lol je l'ai pas

Furtif, 2023-06-02:
vais reload voir ;)

Furtif, 2023-06-02:
dailleurs sa merite une grosse nettoyage 
... comme je disais ... laisser actifs (ne sont pas proprement issues mais bon... ne pas laisser user pendus...

Furtif, 2023-06-02:
> En effet, le probleme est que j'ai pas le temps de m'en occuper, tu veux un acces a ce repos ? Ca serait cool que quelqu'un fasse un peu de menage :)

déjà peut être mettre bien en haut du readme que un issue n'a rien a voir avec une question e futre le lien discutions ... c'est une idée .... après créer un issue récent "Know issues all questions plesase uses "discutions"" sa réveille ... bon je dis sa ... mas a force j'ai pris de la bouteille quand tas plein a raler veut mieux avoir un discord, une base discussion (ou forum) que tus te futem n'importe quoi dans le code... chose que en lui même est super mais si un arrive vois plein issues a sa première vu se dit .... pute !!!! ce truc il a plein des bugs ... ...

Furtif, 2023-06-02:
si tu veut de l'aide das settings add me em colab pas de bug .. je delete mon fork si je suis colab je n'ai plus besoin de fork je work direct

Furtif, 2023-06-02:
Before you close this, see if you change the problem (this is your problem) to https://github.com/cedricp/ddt4all/discussions, you hit something we know I can even give you a *.bat conditional, understand that nothing is done by chance, I said you can drive with your T4 BM etc the thing is generated by the car radio the configuration and you can remove the car's bonnet stripped it's the same if it's not ON it doesn't work and if it's dual check it won't have to be change the geographic ZONE

Furtif, 2023-06-02:
if work ok if not .... need change MAP ############### as can2

Furtif, 2023-06-02:
You can see that this was not coming .... Dexter as this is recent see if it opens a topic in discussions this is closed (even if it is active) the link for this is always active . https://github.com/cedricp/ddt4all/discussions .... happy and check it out ... in fact, I'm going to test it as proposed .

I'm sorry but I'm going to close open topic in discussions tag me if necessary I'll see
I do general cleaning...

closed 2023-06-02 by Furtif

Furtif, 2023-06-03:
https://github.com/cedricp/ddt4all/discussions/841#discussion-5260998

KarelSvo, 2023-06-04:
>As mentioned, I can set the time, but nothing happens. Settings can only be changed with the engine running. Thank you for your time, testing this "problem". It is a useful function during the winter.

Now if the BCM says and -5°C I don't do a remote launch, you have to test on Antarctica.
If the European Union forbids bypassing the immobilizer, renault must prevent this in the BCM firmware ;-)
[image]

Furtif, 2023-06-04:
> Now if the BCM says and -5°C I don't do a remote launch, you have to test on Antarctica. If the European Union forbids bypassing the immobilizer, renault must prevent this in the BCM firmware ;-) [...]

better this https://github.com/cedricp/ddt4all/discussions/841 issue is closed ...

dexter2111, 2023-06-04:
@KarelSvo 
I also tried in winter, with temperatures below -5 degrees.

[4 more replies]