cedricp/ddt4all #644: Registry WINCE MNE

Adame-kr, 2021-12-03:
Hello, everyone.
Does anyone have the option to make and send me the WINCE Export Registry from Media Nav Evolution 9.1.3 ?
Thank you

KarelSvo, 2021-12-04:
> DDT4ALL does not want to make any writing to ECU Audio

https://github.com/cedricp/ddt4all/issues/34

Adame-kr, 2021-12-04:
I've got the cable adjusted. I can see the ECU parameters. But you can't change them.
DDT4ALL writes :
Bad ELM response : WRONG RESPONSE : NR: SubFunction Not Supported(7F2212)

KarelSvo, 2021-12-04:
This error message only relates to individual parameters that are marked  on the screen with red no data.
If you do not show your ddt4all log files, no external error analysis can be carried out.

Adame-kr, 2021-12-04:
Hard reset is the only command to execute.
[elm_ddt.log](https://github.com/cedricp/ddt4all/files/7654691/elm_ddt.log)
[ecu_ddt.log](https://github.com/cedricp/ddt4all/files/7654692/ecu_ddt.log)
[screens.txt](https://github.com/cedricp/ddt4all/files/7654693/screens.txt)

KarelSvo, 2021-12-04:
Could be a capacity Elm problem as the Media Nav Ecu is identified but no acoustic parameters are read. In the log you cannot see whether the Elm also has problems with short data strings. Example menu 11 - Ecu configuration.
You only opened menus with very long data strings.

Adame-kr, 2021-12-05:
ELM test result

v1.0 ATZ [OK]
v1.0 ATWS [OK]
v1.0 ATE1 [OK]
v1.0 ATL1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.0 ATRV [OK]
v1.3 ATS1 [OK]
v1.3 ATS0 [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC0 [OK]
v1.0 ATH0 [OK]
v1.0 ATR1 [OK]
v1.2 ATKW0 [OK]
v1.3 ATD1 [OK]
v1.3 ATD0 [OK]
v1.0 ATAL [OK]
V1.0 ATSP 3 [OK]
V1.0 ATSP 4 [OK]
v1.2 ATIIA 01 [OK]
v1.4 ATSI [OK]
V1.0 ATSP 5 [OK]
v1.4 ATFI [OK]
V1.0 ATSP 6 [OK]
V1.0 ATSP 7 [OK]
V1.0 ATSP 8 [OK]
v1.0 ATPC [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATIB 10 [OK]
v1.0 ATST FF [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SM 1 [OK]
v1.2 ATAT0 [OK]
v1.2 ATAT1 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
v1.4 ATMA [OK/TIMEOUT]
STN11xx TEST 1 STI [FAIL]
STN11xx TEST 2 STDI [FAIL]
STN11xx TEST 3 STPR [FAIL]
STN11xx TEST 4 STPBRR [FAIL]
STN11xx TEST 5 STPRS [FAIL]
Result: 47 succeeded from 52
[elm_ddt_test.txt](https://github.com/cedricp/ddt4all/files/7655511/elm_ddt_test.txt)

Adame-kr, 2021-12-05:
Výsledek menu ECU-configuration

[ecu_ddt.log](https://github.com/cedricp/ddt4all/files/7655520/ecu_ddt.log)
[elm_ddt.log](https://github.com/cedricp/ddt4all/files/7655521/elm_ddt.log)

KarelSvo, 2021-12-05:
Busy Repeat Request Message when writing to the ECU 
Does the media nav start normally or are error messages displayed on the screen?

KarelSvo, 2021-12-05:
There was a repair offer https://github.com/cedricp/ddt4all/issues/113#issuecomment-927524053

Adame-kr, 2021-12-05:
> Zaneprázdněná zpráva s opakovaným požadavkem při zápisu do ECU Spouští se media nav normálně nebo se na obrazovce zobrazují chybové zprávy?

> Busy Repeat Request Message when writing to the ECU Does the media nav start normally or are error messages displayed on the screen?

Media Nav will not be launched at all. Only the logo blinks. Starting in win_ce mode.

Adame-kr, 2021-12-05:
When I compare registry contents [HKEY_LOCAL_MACHINE\LGE\SystemInfo]
here:
 https://renaultlogan2.ru/forum/showthread.php?t=4389&page=30

So the contents of my register are much shorter.
So I would like to compare the status from a working radium of the same version.

Adame-kr, 2021-12-05:
> There was a repair offer [#113 (comment)](https://github.com/cedricp/ddt4all/issues/113#issuecomment-927524053)

The cost of repair and the cost of transportation back and forth is greater than the cost of radium :-((

AndyIreland, 2021-12-05:
I think there was a file or files that can be put on USB stick ... something like ULC recovery or DBOOT .. try looking for them it might repair your medianav evolution

Adame-kr, 2021-12-05:
ULC recovery, DBOOT got. But Media Nav can't fix it.

I wanted to see the WinCe registries from the working radio, but I couldn't find them anywhere.

AndyIreland, 2021-12-05:
i wonder if a dump in pyren can be made from a working Medianav -  and then rollback on your radio using  pyren? - just a thought , although the VIN would not match but maybe that could be changed with PyREn or DDT4ALL ?

Adame-kr, 2021-12-05:
I don't know that. But DDT4all refuses to write. I think he's jotting down data on some database, and it's probably missing from my radio. DUMP from a working radium is not a problem , PYREN does it automatically.

KarelSvo, 2021-12-05:
> i wonder if a dump in pyren can be made from a working Medianav - and then rollback on your radio using pyren? - just a thought , although the VIN would not match but maybe that could be changed with PyREn or DDT4ALL ?

No, internal intervention is necessary in this state. or scrap

AndyIreland, 2021-12-05:
> I don't know that. But DDT4all refuses to write. I think he's jotting down data on some database, and it's probably missing from my radio. DUMP from a working radium is not a problem , PYREN does it automatically.

it does yes. PyREN saves dump files as you use it (if the dump box is ticked)

Adame-kr, 2021-12-05:
> 

What is internal intervention?

Adame-kr, 2021-12-05:
What is internal intervention?

AndyIreland, 2021-12-05:
> What is internal intervention?

I don't know , never heard of it

Adame-kr, 2021-12-05:
> > What is internal intervention?
> 
> I don't know , never heard of it

Neither do I.

AndyIreland, 2021-12-05:
> > > What is internal intervention?
> > 
> > 
> > I don't know , never heard of it
> 
> Neither do I.

I think it means though maybe a 'flasher' put inside onto (TP) test pin spots and flash firmware straight into the ROM chip

AndyIreland, 2021-12-05:
> >  Media Nav will not be launched at all. Only the logo blinks. Starting in win_ce mode.

Well I tell you what , to me that sounds pretty positive that you have actually got a logo blinking there - I would be less hopeful if that logo never come up and the screen was black then I would call that totally 'bricked' ... but with a Logo coming up it sounds only 'partially bricked' .. so I would have thought there is hope there somewhere?

Adame-kr, 2021-12-06:
> Well I tell you what , to me that sounds pretty positive that you have actually got a logo blinking there - I would be less hopeful if that logo never come up and the screen was black then I would [...]

I also think there's hope. It's just a special computer with a WINCE system. If I can get to the windows, there's got to be a way to save it. I would therefore like to compare the contents of the register. I need to find someone who has the same Media Nav and will export the registry to me using Regedit.exe . I found something online, but it's for an old MN ver 4.0.6. But I need a newer version for MN Evolution.

AndyIreland, 2021-12-06:
I will try to get a chance today to see if it can be backed up on my wife's Clio Mk4 with Medianav 9.1.3 @Adame-kr - and i put medianav 5.0 plus on it about a year ago so that might have a registry backup icon on it - 
http://www.mondoduster.it/2018/12/il-nuovo-plus-v5-0-solo-per-evolution/

Adame-kr, 2021-12-06:
> I will try to get a chance today to see if it can be backed up on my wife's Clio Mk4 with Medianav 9.1.3 @Adame-kr - and i put medianav 5.0 plus on it about a year ago so that might have a registry backup icon on it - http://www.mondoduster.it/2018/12/il-nuovo-plus-v5-0-solo-per-evolution/

Thank you. I think it will help me. Maybe I'll fix it, or I'll know the registers are a dead end.

AndyIreland, 2021-12-06:
can you get to see the file manager through wince on the medianav? - are all the storage folders there (I think there are 4 of them)

Adame-kr, 2021-12-06:
Yes. I see all four of them. I also have their contents for version 9.1.3
It's all easy to find on the Internet.
I also have a backup of my own license.

AndyIreland, 2021-12-06:
here is the reg export from the wife's clio mk4 medianav evo if its what you are looking for:

https://we.tl/t-IYIEOqqb9r

Adame-kr, 2021-12-06:
> here is the reg export from the wife's clio mk4 medianav evo if its what you are looking for:
> 
> https://we.tl/t-IYIEOqqb9r

Thank you. I'm going to start experimenting.

AlMeu, 2022-01-19:
Still not fix the MediaNav?

closed 2023-06-02 by Furtif