cedricp/ddt4all #645: Whenever I try to start in "connected mode" the program just crashes

cvjensen, 2021-12-05:
Hello , 

I have installed DDT4ALL using the fully packaged installer for windows.

I'm using an ELM327 BT device that should have the right PIC18F25K80 chip .. 
I opened it up and there's the right number on the chip but I still dont know if it's a chinese copy?

I have tested the OBD BT device using my phone and some android apps - seems to be working fine.

If I dont start a command prompt first I cannot manage to read the errors, so here's the output from the command prompt whenever I try to press the "connected mode" button:

```
C:\Users\Rikke\Desktop>.\ddt4all.lnk
Initilizing ELM with speed 115200...
Trying to open portCOM3 @ 115200
Traceback (most recent call last):
  File "C:\Program Files (x86)\ddt4all\ddt4all.py", line 1467, in <module>
    options.elm = elm.ELM(options.port, port_speed, pc.adapter, pc.raise_port_speed)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 574, in _init_
    maxspeed = int(maxspeed)
ValueError: invalid literal for int() with base 10: ''
ELM reset...

C:\Users\Rikke\Desktop>.\ddt4all.lnk
Initilizing ELM with speed 115200...
Trying to open portCOM4 @ 115200
Traceback (most recent call last):
  File "C:\Program Files (x86)\ddt4all\ddt4all.py", line 1467, in <module>
    options.elm = elm.ELM(options.port, port_speed, pc.adapter, pc.raise_port_speed)
  File "C:\Program Files (x86)\ddt4all\elm.py", line 574, in _init_
    maxspeed = int(maxspeed)
ValueError: invalid literal for int() with base 10: ''
ELM reset...
```

Notice I do it 2 times - thats because for some reason the BT device shows up on windows 2 times. When I connect to both of the BT OBDII devices I get 2 different COM-ports - thats a little strange isn't it ?

cvjensen, 2021-12-05:
Found this online:

`The reason you had ValueError is because int cannot convert an empty string to the integer.`

But it clearly states that it's trying with speed 115200 which I'm guessing is default for BT?

KarelSvo, 2021-12-05:
maxspeed is a function of the Python 3 version. The Python 3 version still contains some bugs. I recommend using the Python 2 version.

cvjensen, 2021-12-05:
I will try that one of the following days. It's just funny that the readme says that Python 3.8 is a dependency .

AndyIreland, 2021-12-05:
have you tried downloading DDT4ALL from this Russian site , choose 'all cars exe'  - and try choose 34800 baud rate - 

https://www.drive2.com/b/498093336985338243/

yes that is normal 2 ports with a BT Dongle

cvjensen, 2021-12-06:
@AndyIreland seems to be working fine. It is just great , thanks!

closed 2021-12-06 by cvjensen

cvjensen, 2021-12-23:
hi @AndyIreland , you don't by any chance know how to calibrate an edc gearbox ? :)

AndyIreland, 2021-12-23:
Hiya - no, i have only made adjustments on Clio4 (non transmission settings) and Zoe EV sorry. 
could it be in the ECM / injection ECU section or that may be just be for the engine parameters?