cedricp/ddt4all #650: Syntax error in DDT4ALL.PY

AndyIreland, 2021-12-26:
Hello can someone tell me the problem here please? - 

I have downloaded ddt4all.py from github here and renamed my existing ddt4all.py to ddt4all.pyold in the program files (386) folder and now when i start up DDT4all from the desktop shortcut I get a syntax error and cannot go any further.

Thanks:

[image]

KarelSvo, 2021-12-27:
A good example of, what mixing different versions of files does ;-)

AndyIreland, 2021-12-27:
> A good example of, what mixing different versions of files does ;-)

and ... - any solution? :)

KarelSvo, 2021-12-27:
well just don't throw files from different versions together  :D

AndyIreland, 2021-12-29:
OK , so I just tried downloading from github @cedricp & @KarelSvo  and I cannot get DDT4ALL to start I just get this when I press ddt4all.bat  then press any key to continue

[image]

 ... but the screen goes away and DDT4all will not start. I have put the ecu, vehicles, and python38 folders in the right place what am I doing wrong with the official download from github?

KarelSvo, 2021-12-29:
https://github.com/cedricp/ddt4all/releases

AndyIreland, 2021-12-29:
> https://github.com/cedricp/ddt4all/releases

Thanks @KarelSvo - i found the installer file - installed it , copied my large ecu folder of 3gb to ddt4all folder in program files (386) and DDT4all starts up fine now but choose edition mode (demo) and it says 1 ecu found ... any ideas what is going on there and why it hasnt loaded the ecu's ? thanks.

KarelSvo, 2021-12-30:
I have already written very often that I am not using this Python 3 version because it still contains some bugs.
What exactly your problem is cannot be deduced from your description. 
An uncompressed ECU database must contain a suitable ECU list so that the database can be loaded.
With the database compression function built into ddt4all, a suitable ECU list is created automatically.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 need first about a litle know prog for it....

closed 2023-06-02 by Furtif