cedricp/ddt4all #651: Renault grand scenic 2 ddt4all

Wowa2507, 2022-01-02:
Renault grand scenic 2.  Good day, everyone needs your help, there are 2 ELM 327, one version is 2.1 and the other is 1.5 for 2 boards.  When I insert 2.1 bluetooth, then through ddt4all the connection. And version 1.5 for 2 boards is not connected.  there is Windows 10 x32.  I want to register a key card. Elm version 2.1 does not show the isk pin.  auto 2007
#

amaurylopin, 2022-01-03:
Hi,

It might be your ELM that doesn't support DDT4ALL. 

You might want to have a look at this kind of elm [https://www.amazon.fr/ScanTool-425801-obdlink-SX-USB-professionnel/dp/B005ZWM0R4](https://www.amazon.fr/ScanTool-425801-obdlink-SX-USB-professionnel/dp/B005ZWM0R4)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 aucune cuille ....

Furtif, 2023-06-02:
usb est dans le read me mis  par que a un derleck dedier

closed 2023-06-02 by Furtif