cedricp/ddt4all #652: Can't connect to ELM327 in BT with ddt4all

Greby, 2022-01-09:
Hi everyone!

I have a problem with my ELM327 (PIC18F25K80 Chip). So I try to use ddt4all
I can connect to the OBD2 in Bluetooth from my laptop. (Win7 x64)
 I tried with the firewall and anti-virus disabled but it didn't help.
When I click on "Tests ELM" , I get this message:

v1.0 ATZ [OK]
v1.0 ATWS [OK]
v1.0 ATE1 [OK]
v1.0 ATL1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.0 ATRV [OK]
v1.3 ATS1 [OK]
v1.3 ATS0 [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC0 [OK]
v1.0 ATH0 [OK]
v1.0 ATR1 [OK]
v1.2 ATKW0 [OK]
v1.3 ATD1 [OK]
v1.3 ATD0 [OK]
v1.0 ATAL [OK]
V1.0 ATSP 3 [OK]
V1.0 ATSP 4 [OK]
v1.2 ATIIA 01 [OK]
v1.4 ATSI [OK]
V1.0 ATSP 5 [OK]
v1.4 ATFI [SIKERTELEN]  //NOK
V1.0 ATSP 6 [OK]
V1.0 ATSP 7 [OK]
V1.0 ATSP 8 [OK]
v1.0 ATPC [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATIB 10 [OK]
v1.0 ATST FF [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SM 1 [OK]
v1.2 ATAT0 [OK]
v1.2 ATAT1 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
Tracback (most recent call last)
 File ".\ddt4all.py", line 1201, in check_elm
  res = elm.elm_checker(port, speed, self.logview, coreQcoreApplication)
 File "C:\Program Files (x86)\ddt4all\elm.py", line 1328, in elm_cheker
  res = elm.send_raw(cm[2])
 File "C:\Program Files (x86)\ddt4all\elm.py", line 1063, in send_raw
  self.lf.write("<[" + str(round(tc - tb, 3)) + "]" + self.buff + "\n")
UnicodeEncodeError: 'ascii' codec can't encode character u'\u0150' in position 33: ordinal not in range(128)

Maybe my OBD2 is not compatible?
Any other idea?

Thanks!

AndyIreland, 2022-01-10:
try uninstalling your version of DDT4ALL and then downloading this version from here and choose "ALL CARS EXE" and install it see if that works:
https://www.drive2.com/b/498093336985338243/

KarelSvo, 2022-01-10:
show the file elm_ddt

Greby, 2022-01-11:
> try uninstalling your version of DDT4ALL and then downloading this version from here and choose "ALL CARS EXE" and install it see if that works: https://www.drive2.com/b/498093336985338243/

I've tried, same problem

Greby, 2022-01-11:
> show the file elm_ddt

>[08:26:40.415]ATZ
<[0.125]ATZ

ELM327 v1.5

>
>[08:26:40.540] ATZ                   
<[0.125] ATZ                   

ELM327 v1.5

>
>[08:26:40.696] ATWS                  
<[0.016] ATWS                  

ELM327 v1.5

>
>[08:26:40.727] ATE1                  
<[0.0] ATE1                  
OK

>
>[08:26:40.727] ATL1                  
<[0.0] ATL1                  
OK

>
>[08:26:40.743] ATL0                  
<[0.031] ATL0                  

OK

>
>[08:26:40.774] ATI                   
<[0.016] ATI                   
ELM327 v1.5

>
>[08:26:40.790] AT@1                  
<[0.031] AT@1                  
OBDII to RS232 Interpreter

>
>[08:26:40.821] ATRV                  
<[0.015] ATRV                  
12.2V

>
>[08:26:40.852] ATS1                  
<[0.0] ATS1                  
OK

>
>[08:26:40.868] ATS0                  
<[0.031] ATS0                  
OK

>
>[08:26:40.899] ATCAF0                
<[0.015] ATCAF0                
OK

>
>[08:26:40.914] ATCFC0                
<[0.032] ATCFC0                
OK

[... 467 lines cut ...]
<[0.0]AT CRA 765
OK

>
>[08:27:59.208]AT FC SH 745
<[0.015]AT FC SH 745
OK

>
>[08:27:59.223]AT FC SD 30 00 00
<[0.016]AT FC SD 30 00 00
OK

>
>[08:27:59.239]AT FC SM 1
<[0.0]AT FC SM 1
OK

>
>[08:27:59.239]AT SP 6
<[0.0]AT SP 6
OK

>
>[08:27:59.239]0210C0

KarelSvo, 2022-01-13:
Remove the log file and do an Elm test with the software from the link.
show the new log data

Greby, 2022-01-17:
> Remove the log file and do an Elm test with the software from the link. show the new log data I want to see test result from atpps.

The test results:
>[10:36:50.980]ATZ
<[0.11]ATZ

ELM327 v1.5

>
>[10:36:51.090] ATZ                   
<[0.093] ATZ                   

ELM327 v1.5

>
>[10:36:51.277] ATWS                  
<[0.062] ATWS                  

ELM327 v1.5

>
>[10:36:51.339] ATE1                  
<[0.016] ATE1                  
OK

>
>[10:36:51.355] ATL1                  
<[0.031] ATL1                  
OK

>
>[10:36:51.386] ATL0                  
<[0.016] ATL0                  

OK

>
>[10:36:51.417] ATI                   
<[0.047] ATI                   
ELM327 v1.5

>
>[10:36:51.464] AT@1                  
<[0.0] AT@1                  
OBDII to RS232 Interpreter

>
>[10:36:51.526] ATRV                  
<[0.016] ATRV                  
12.3V

>
>[10:36:51.542] ATS1                  
<[0.016] ATS1                  
OK

>
>[10:36:51.558] ATS0                  
<[0.031] ATS0                  
OK

>
>[10:36:51.589] ATCAF0                
<[0.015] ATCAF0                
OK

>
>[10:36:51.620] ATCFC0                
<[0.016] ATCFC0                
OK

>
>[10:36:51.636] ATH0                  
<[0.031] ATH0                  
OK

>
>[10:36:51.667] ATR1                  
<[0.015] ATR1                  
OK

>
>[10:36:51.682] ATKW0                 
<[0.047] ATKW0                 
1:-- 2:-->
>[10:36:51.729] ATD1                  
<[0.031] ATD1                  
OK

>
>[10:36:51.760] ATD0                  
<[0.032] ATD0                  
OK

>
>[10:36:51.792] ATAL                  
<[0.015] ATAL                  
OK

>
>[10:36:51.807] ATSP 3                
<[0.031] ATSP 3                
OK

>
>[10:36:51.838] ATZ              
<[0.125] ATZ              

ELM327 v1.5

>
>[10:36:51.963] ATSP 4                
<[0.0] ATSP 4                
OK

>
>[10:36:51.963] ATIIA 01              
<[0.016] ATIIA 01              
OK

>
>[10:36:51.994] ATSI                  
<[0.016] ATSI                  
OK

>
>[10:36:52.026] ATZ              
<[0.109] ATZ              

ELM327 v1.5

>
>[10:36:52.135] ATSP 5                
<[0.0] ATSP 5                
OK

>
>[10:36:52.135] ATFI                  
<[0.031] ATFI                  
?

>
>[10:36:52.166] ATZ              
<[0.125] ATZ              

ELM327 v1.5

>
>[10:36:52.291] ATSP 6                
<[0.0] ATSP 6                
OK

>
>[10:36:52.291] ATSP 7                
<[0.015] ATSP 7                
OK

>
>[10:36:52.306] ATSP 8                
<[0.032] ATSP 8                
OK

>
>[10:36:52.338] ATPC                  
<[0.015] ATPC                  
OK

>
>[10:36:52.369] ATSH 012              
<[0.015] ATSH 012              
OK

>
>[10:36:52.384] ATSW 96               
<[0.032] ATSW 96               
OK

>
>[10:36:52.416] ATSW 34               
<[0.015] ATSW 34               
OK

>
>[10:36:52.447] ATWM 817AF13E         
<[0.0] ATWM 817AF13E         
OK

>
>[10:36:52.462] ATIB 10               
<[0.032] ATIB 10               
KBusBaud=10400 OK

>
>[10:36:52.494] ATST FF               
<[0.031] ATST FF               
OK

>
>[10:36:52.540] ATCRA 012             
<[0.016] ATCRA 012             
OK

>
>[10:36:52.556] ATCRA 01234567        
<[0.031] ATCRA 01234567        
OK

>
>[10:36:52.587] ATFC SH 012           
<[0.016] ATFC SH 012           
OK

>
>[10:36:52.618] ATFC SH 00112233      
<[0.016] ATFC SH 00112233      
OK

>
>[10:36:52.634] ATFC SD 300000        
<[0.031] ATFC SD 300000        
OK

>
>[10:36:52.665] ATFC SM 1             
<[0.016] ATFC SM 1             
OK

>
>[10:36:52.681]ATAT0                  
<[0.0]ATAT0                  
OK

>
>[10:36:52.696]ATAT1                  
<[0.032]ATAT1                  
OK

>
>[10:36:52.743] ATCM 123              
<[0.0] ATCM 123              
OK

>
>[10:36:52.743] ATCM 12345678         
<[0.016] ATCM 12345678         
OK

>
>[10:36:52.759] ATCF 123              
<[0.031] ATCF 123              
OK

>
>[10:36:52.806] ATCF 12345678         
<[0.031] ATCF 12345678         
OK

>
>[10:36:52.837] ATZ              
<[0.125] ATZ              

ELM327 v1.5

>
>[10:36:52.962] ATMA

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 relatif...

closed 2023-06-02 by Furtif