cedricp/ddt4all #654: ddt4all 2.0 Release November 2020 Python 3 not starting

AlMeu, 2022-01-15:
I tried and install under Win7 x32 like old version also this new version.
But when I try run it I get this error:

[image]

What I miss or I am doing wrong?

AndyIreland, 2022-01-15:
i dunno - but try downloading DDT4ALL from here and choose 'all Cars' 
https://www.drive2.com/b/498093336985338243/

AlMeu, 2022-01-19:
@AndyIreland 
If I want test new version, for what I have to download old version?
I am interested to see what's new in this version, but it starts with wrong foot.

AndyIreland, 2022-01-19:
I have tried both but I find https://www.drive2.com/b/498093336985338243/ all cars more reliable/stable

KarelSvo, 2022-01-20:
https://www.python.org/downloads/windows/

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 need py 3.10.x

closed 2023-06-02 by Furtif