cedricp/ddt4all #658: Megane 4 daylight DRL intensity...

JohnMoro86, 2022-01-28:
Hi everyone, on Megane IV I would like to know if it is possible to change the brightness of the daytime LEDs in the EEM control unit, during the day. 

I doubt that they are not set to the maximum since they are not visible with the sun. I was able to find the intensity when it is night with the low beam on but I can't find the line for the day. 

Thanks to those who can help me, I will be grateful to you.

AndyIreland, 2022-01-28:
you could try changing this to '1' its DRL without PWM  .....but it would mean that DRL's are at full brightness all the time even with headlamps on , because PWM is pulse width modulation (which dims LED's) 

[image]

JohnMoro86, 2022-01-28:
Thanks for the help, Ok tonight I will try and see at least if the brightness is now at maximum, so there is no single parameter for the brightness only during the day

JohnMoro86, 2022-01-28:
Thank you very much ;), I'll try to see these parameters and update you

JohnMoro86, 2022-01-30:
[image]

I tried to change the parameters, they were set as you see at 0.6, it seems that increasing the value increases the intensity but I'm not sure. 

I had the impression that raising close to the value 1, that is I believe the maximum value, the led strip seems to become more yellow due to the higher intensity I believe.

AndyIreland, 2022-01-30:
The LED strips for DRLs should always be cool white colour rendering - 0.6 I would say is dim to 60% brightness and '1' would be full brightness - should not turn them yellow I dont think.

JohnMoro86, 2022-01-30:
Now I have set it to 0.75 it seems to me a good intensity, much better than before. 

I do not understand why Renault sets it to 60%, I just hope that in the long run, increasing intensity, it does not give the known problems on many Megane of the led light that turns yellow because the led guide is ruined.

AndyIreland, 2022-01-30:
> Now I have set it to 0.75 it seems to me a good intensity, much better than before.
>
> I do not understand why Renault sets it to 60%, I just hope that in the long run, increasing intensity, it does [...]

Ah right I had not heard of that - I have heard of UV sun rays turning the clear plastic lens of DRL's and headlights yellow - if you are worried put it back to 0.6 

What happened when you adjusted the 'DRL without PWM' setting - did they go full brightness for you?

JohnMoro86, 2022-01-30:
DRL without PWM I have not changed it because it was already set to 1 (so I think the maximum), now I have put the other 2 values ​​at 0.75 and it seems to be more visible with the sun without exaggerating with the brightness, it seems to me the right compromise for now.  As for the deterioration of the guide, I think it is the fault of the sun or the halogen bulb that heats the headlight too much, I think that the 3 LEDs that make light in the guides are unable to create so much heat to deteriorate the headlight, even at maximum power.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 I found out the same

closed 2023-06-02 by Furtif