cedricp/ddt4all #661: [Scenic IV] how to activate heated and electric seats?

szeryf85, 2022-02-09:
Hi :)
I bought heated and electrically operated seats. I made a wiring harness and connected the seats to the car's electrical system. The problem is that the CAN Clip is not detecting the modules in the seats. Is there any way to activate this in ddt4all?

Regards,
Lukas

AndyIreland, 2022-02-09:
Have you enabled them in the infotainment / HMI system of your car? - this is for Scenic 4 phase 1 :

[image]

szeryf85, 2022-02-09:
@AndyIreland thank you for your response.
after entering the module settings there was something like this:
[image]

after changing to "with heating" and trying to save I have this error:
[image]
Request aborted (look at red paramters entries): NO DATA

and this is the only line with the entry "NO DATA"
[image]

was there a connection problem? the interface is ok, because it connects to other modules.

AndyIreland, 2022-02-09:
OK , so you have the specially wired ELM327 or soldered lead so that you can read the radio/infotainment system through the CAN socket yes?
Other than that , try changing the CAN timeout to 600ms and hit the refresh button and make sure the red No data has gone

szeryf85, 2022-02-09:
I have an ELM cable with a switch between CAN_H (6), CAN_L (14) and CAN_H (13), CAN_L (12) - low speed CAN. Which CAN line should I choose for this connection? I tried both, and it didn't work. I will increase the CAN timeout to 600ms.

AndyIreland, 2022-02-09:
This wiring diagram should help out and try make sure you are using an ELM327 with a PIC18F25K80 chip in it too if you can
[image]
https://www.gps-rlink.com/r-link-1/activate-android-auto-waze-on-r-link-1/

szeryf85, 2022-02-09:
ok, so should I choose CAN_H (13), CAN_L (12) for this connection?

AndyIreland, 2022-02-09:
I *think* its CAN H for reading the radio / infotainment system

szeryf85, 2022-02-09:
still the same error. I checked and I have ELM327 with a PIC18F25K80 chip.

AndyIreland, 2022-02-10:
shame that - and you done a search for the ecu's in your car using the magnifying glass in ddt4all did you and see what come, not just pick the ecu manually?

szeryf85, 2022-02-11:
I tried searching but it did not find this module. only that it connects easily with some modules that it does not find. I'm looking for a different Elm327 interface.

BarektE, 2022-06-02:
Of course you need to activate 1. From UCH 2. From TDB

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 rapport it to diccussions

closed 2023-06-02 by Furtif