cedricp/ddt4all #662: Always dpf regeneration and high fuel consumption

flame00111, 2022-02-09:
Hi guys my car is 2014 clio iv 1.5 dci edc 90 hp. My problem is in city druve high fuel consumption very slowy drive and it say 7.5 L / 100 km. When i open pyren or torque i see my car always when i drive dpf regeneration. My soot 6.7 gr 6.8 gr and i still drive in city its slowy down 6.6  6.5 6.4 gr. Its when i drive always regeneration. I drive 45 L diesel 600 km. And 7.5 Liter fuel consumption. I look forums its high to clio people says its consumption 5.5 - 6 Liters in city drive. And my engine doesnt hot its always 75 -77 degree. Not to come up 88 - 90 degrees. Help please

flame00111, 2022-02-10:
Anyone ?

Jodaille, 2022-08-16:
far from being an expert and not mechanic, if "dpf regeneration" is the right diagnostic, maybe you have to have a tour and go high on rpm to help it burning particules.

erhsntrk, 2022-10-08:
Have the water thermostat checked. Since it is always open, the water is constantly circulating and the engine does not warm up.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif