cedricp/ddt4all #667: OBDLink EX USB instead of SX

primozkikelj, 2022-02-22:
Hi!

I am planning to buy OBDLink EX USB for some other applications, and would like to know if ddt4all works with this adapter also?

Thanks for your answer!

Regards,

Primož

DM-2015, 2022-03-12:
Hello,

I bought OBDLink EX last week and I have a communication problem. I have the message "No ELM or no OBDLink SX not recognized COM5". My laptop has port USB 3.0.

I looking for help.

Greetings

AndyIreland, 2022-03-12:
try plugging into a non USB 3 port. on laptops normally you have a standard USB socket or 2 . You may not have the required USB 3.0 installed on your laptop

KarelSvo, 2022-03-13:
> Hello,
> 
> I bought OBDLink EX last week and I have a communication problem. I have the message "No ELM or no OBDLink SX not recognized COM5". My laptop has port USB 3.0.
> 

https://www.scantool.net/scantool/downloads/archive/
In the utilities area there is a finder and a terminal software that you can use to test the basic operational readiness of your obdlink.
If the test is OK, the problem lies with ddt4all.

EXcomZkko, 2022-03-23:
Check the baudrate via st term utility.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 .... but is in readme....

closed 2023-06-02 by Furtif