cedricp/ddt4all #670: Parking sensors

thomson56, 2022-03-03:
Hello, I need the settings in ddt4all for parking sensors and park asist for kadjar, I had a little fun and can't activate them back

AndyIreland, 2022-03-03:
thats why important to make a dump txt file backup in pyREN first of all before changing any parameters  with DDT4ALL :)

thomson56, 2022-03-04:
It's all great just for me to know how to do it

luis8097, 2022-03-25:
@AndyIreland explain how to make dump file in Pyren

KarelSvo, 2022-03-25:
The dump will be created automatically, if this option is enabled at startup.  https://gitlab.com/py_ren/pyren/-/wikis/Using%20launcher

munkun46, 2022-12-23:
How can I activate my car's front parking sensors with ddt4all?  It doesn't work when I try, the values ​​don't change. Kadjar 1.2 Tce, thaks
[image]

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Kaszanek1018, 2025-05-16:
Hi, can someone please share settings for parking sensors for Renault Talisman. I also had a bit too much fun and didnt know about any dump txt backup files. 

Renault talisman 2017 Intens Energy dCi 160 EDC