cedricp/ddt4all #672: Replacing rlink module with brand new one

yagokhan, 2022-03-06:
I decided to change the rlink unit with brand new one as mine is ver 2.2 and no carplay. My question is do i need to write the vin of my car to the new unit? And will i be able to write it using dd4all or something else?
Thanks for help.

AndyIreland, 2022-03-06:
I should imagine you need to write VIN and CRC of your car (DDT4ALL does have CRC calculator utility  built in) so i reckon you do need to put in VIN and CRC - your next issue though could be the anti-Theft thing, I think they add it to the rlink 1 and am presuming they add it to the  rlink 2 unit as well. - i dont know how you do that

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif