cedricp/ddt4all #673: R-LINK2 WideBandSpeech - anyone tried to set it up?

jano7878, 2022-03-19:
Hi, I have Koleos with R-LINK 2 HW Gen 2.1 and V9 9.0.35.506

This kit has terrible bluetooth hands free call quality, it supports NarrowBand speech, without Widenband.
Wideband is common in current smartphones. I had found WideBandSpeech parameter on RadNav 3.0 configuration. It was set to Not Active so I changed it to Active, but without any positive results.  Calls via bluetooth are still narrowband. 

Anyone has idea how to turn R-Link2 WidebandSpeech on? Maybe any additional configuration changes are required?
I observed, that on EasyLink configuration A-IVI (which fully supports WidebandSpeech via bluetooth phone) there are separate Audio configurations for Narrowband (Phone_EC/NR_NBS) and for Wideband (Phone_EC/NR_WBS). So EasyLink HF calls are crystal clear because of full Wideband support.

Maybe any changes in RadNav Phone Acoustic Parameters are also required to turn R-Link2 WideBandSpeech on?

Thanks in advance for any help! I really hate to use R-Link 2 bluetooth HF because of voice quality like in old Nokia cellular.

sajadkhanizade64, 2022-04-17:
Send me a pic from ur rlink; radnav 3.and radnav 3x acustic page

jano7878, 2022-04-27:
I attached Acoustic Parameters, Phone Parameters, RadNav3.0 files - screens from RadNav 3.X v3.3
Additionally, I have attached the file RadNav 3.0 - the screen from RadNav 2.4
In both cases I have the same problem: trying to read long parameters generates an error [BAD RESPONSE]

[image]

[image]
[image]
[image]

Fatih9871, 2023-05-19:
Radnav 3.3 parametre olan varmi