cedricp/ddt4all #675: Wifi connection problem Winerror 10053

joaomolina1, 2022-04-01:
Hey there. 

So i have a wifi ELM327 with the chip PIC18F25K80 and i get no problems connecting it to my phone. 

But when i try to connect with my pc, DDT4ALL gives me the error "[WinError 10053] An established connection was aborted by the software in your host machine" and dosen't connect. It tries all the speeds until it fully disconnects. 

I am using the most updated software and i disabled all my firewalls and stuff. Any idea how to fix this?

mnvkbvbm, 2022-06-04:
yes, I'm getting the same error. ddt4all v2.0 also gives this error. Older versions don't give this error. I'm getting this error in the latest version.

AndyIreland, 2022-06-04:
what python version is in your ddt4all folder in program files (x86) - try python version 27

mnvkbvbm, 2022-06-04:
python38 is installed. should we delete it and install python27?

AndyIreland, 2022-06-04:
you could try out python27 - you will have to change the line in DDT4ALL.BAT to point it to python27 

start Python27/Python.exe ddt4all.py
@echo off
CLS
ECHO Running "ddt4all.py2 Python scripts...
Python.exe ddt4all.py
ECHO.
ECHO.
pause

KarelSvo, 2022-06-04:
https://github.com/cedricp/ddt4all/pull/659/files

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836