cedricp/ddt4all #676: Megane 4 grandtour

tilenkosec, 2022-04-08:
Hi there,

I was playing around in the developer mode on my radio unit, when I clicked the "High display mode", the display went dark.
After that, I used [Pyren](https://gitlab.com/py_ren/pyren) to solve it and it came back to life but after that, I was having problems with climate buttons.

I tried to fix that but now the whole climate is not working and the screen is black again.

Is anyone here who could guide me through the process to get it back online?

I have an ELM cable and a ddt4all program.

Car:
Megane 4 Grandtour 2017

killuadk, 2022-05-16:
connect with ddt4all >> chose your care >> select audio >> select your rlink2 version  >> 
you will find somme of conf with value search in the net for value like your display  (i think the is just 2 type of value because there are 2 display HIGH and MED )

tilenkosec, 2022-05-16:
> connect with ddt4all >> chose your care >> select audio >> select your rlink2 version >> you will find somme of conf with value search in the net for value like your display (i think the is just 2 type of value because there are 2 display HIGH and MED )

I did manage to get it back online by using Talisman settings, now the issue is that the climate is not working correctly.. when I set 16c+ it heats like crazy, and when it's on low it cools down hard.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif