cedricp/ddt4all #680: PyRen - DdT4All error

org07, 2022-05-06:
Hello everyone, I have a problem, I want support from those who have information. My ELM 327 device works with PYREN, but DDT4ALL gives an OBD com error how can I solve this problem?

AndyIreland, 2022-05-06:
make sure you choose the correct port, and try reducing the baud rate and try reducing the ms CAN timeout settings

KarelSvo, 2022-05-07:
show the error message

org07, 2022-05-07:
It works fine in PyRen, I tried everything in ddt4all, it doesn't work. error message com port not found

AndyIreland, 2022-05-07:
strange . Is it USB , Bluetooth or wiFi ELM327?

AndyIreland, 2022-05-07:
> It works fine in PyRen, I tried everything in ddt4all, it doesn't work. error message com port not found

do you exit fully out of pyREN before trying to connect with DDT4ALL? 

and try this version of DDT4ALL and select 'ALL CARS' 

https://www.drive2.com/b/498093336985338243/

org07, 2022-05-08:
I was already using it all the time, the bluetooth ELM327 I used before is broken, I am having this problem on the device I bought later. The strange thing is that although PyRen and Ddt4all use the same infrastructure, it works in one but not in the other

[image]
.

org07, 2022-05-08:
Let me point out that I am not a newbie in using it.

KarelSvo, 2022-05-08:
WiFi Elm and ddt4all with Python 3 produce a connection error.  https://github.com/cedricp/ddt4all/pull/659/files

org07, 2022-05-08:
> Python 3 ile WiFi Elm ve ddt4all bir bağlantı sorunu olmuyor. https://github.com/cedricp/ddt4all/pull/659/files

the picture shows wifi but the device i bought has bluetooth

KarelSvo, 2022-05-08:
https://github.com/cedricp/ddt4all/pull/679/files

org07, 2022-05-08:
> https://github.com/cedricp/ddt4all/pull/679/files

Mr. Karel Svo this is the problem

[image]

org07, 2022-05-08:
Does anyone know what the source of this problem is?

org07, 2022-05-08:
It also works with Android ECU Tweaker app. Only PC doesn't work with ddt4all. Isn't there a person to help?

AndyIreland, 2022-05-08:
go into add/remove bluetooth device in windows and remove the obd2 adaptor (if it shows up) then go into device manager of windows and delete all ports (if COM ports show up) and remove them , Then go into add remove bluetooth device and re add the bluetooth obdII adaptor and see if that works. If your BT adaptor works on the Android version of DDT4ALL (ECU tweaker) and Pyren I cannot see why it will not work with your windows laptop PC - if it dont then i would say its a problem up the laptop end settings or the laptop's bluetooth adaptor

Furtif, 2022-05-08:
Serial bug if com3 err try https://github.com/cedricp/ddt4all/pull/634

org07, 2022-05-09:
> Windows'ta bluetooth aygıtı ekle/kaldır'a gidin ve obd2 bağdaştırıcısını kaldırın (görünürse), ardından pencerelerin aygıt yöneticisine gidin ve tüm bağlantı noktalarını silin (COM bağlantı noktaları görünüyorsa) ve bunları kaldırın, Ardından Bluetooth aygıtını ekle'ye gidin ve bluetooth obdII adaptörünü yeniden ekleyin ve işe yarayıp yaramadığına bakın. BT adaptörünüz DDT4ALL (ECU tweaker) ve Pyren'in Android sürümünde çalışıyorsa, Windows dizüstü bilgisayarınızla neden çalışmayacağını göremiyorum - çalışmıyorsa, dizüstü bilgisayar uç ayarlarında veya dizüstü bilgisayarın bluetooth'unda bir sorun olduğunu söyleyebilirim. adaptör

Dear Andy Ireland also sees PyRen on the computer. I did all of what you said, but ddt4all doesn't work with ELM327 bluetootht. I believe 2 people will solve this problem here. 1. Cedric PAILLE 2. shrlnm I trust these two so much

org07, 2022-05-09:
> 

> [com3 hatası #](https://github.com/cedricp/ddt4all/pull/634) 634'ü denediğinde seri hata[](https://github.com/cedricp/ddt4all/pull/634)

Dear Furtif, I did not understand anything from this link, but thank you for your interest.

Furtif, 2022-05-09:
@org07 this link is an open pull request its quite simple to use just clone the repo, install pyrhon 3.x and run dtt4all.bat ..

Furtif, 2022-05-09:
if you prefer you can also create an updated setup with that.. your problem and just use the setup content without being interested in anything else .. in open pull there are new cars according to the latest data bases ..

org07, 2022-05-09:
> @org07 this link is an open pull request its quite simple to use just clone the repo, install pyrhon 3.x and run dtt4all.bat ..

I already have Python 3.10.4 installed on my computer.

org07, 2022-05-09:
> @org07 this link is an open pull request its quite simple to use just clone the repo, install pyrhon 3.x and run dtt4all.bat ..

I don't know how to do what you said :(

Furtif, 2022-05-09:
@org07 it's easy if you have git.  installed just open a terminal and run what is in readme otherwise you can download it in zip

Furtif, 2022-05-09:
if all fine ok if not use this zip 
[dtt4all-furtif-branch.zip](https://github.com/cedricp/ddt4all/files/8653022/dtt4all-furtif-branch.zip)
drop this wherever you want to launch ddt4all double click on dtt4all.bat or dtt4all.sh if using linux or mac os

Furtif, 2022-05-09:
if not ok need install venv remove rem of bat file in lines echo install venv etc... or use this
[DDT4ALL.zip](https://github.com/cedricp/ddt4all/files/8653055/DDT4ALL.zip) *.bat ok ... i have zipe mine i have already venv installed ... 
```
@echo off
echo Install new venv ...
python -m venv ./venv
echo Activate venv ...
call venv\Scripts\activate.bat
python -m pip install --upgrade pip
echo Install requirements ...
pip install -r requirements.txt
echo Run's app ...
python ddt4all.py
rem echo Deactivate venv ...
rem call venv\Scripts\deactivate.bat
pause
```
fine

Furtif, 2022-05-09:
@org07 don't forget to copy the database inside as @cedricp mentions in readme file

Furtif, 2022-05-09:
python 3.10.4  maybe get some errs this works fine in python 3.8

org07, 2022-05-09:
Dear Furtif, thank you very much, I'll try it tomorrow and let you know

Furtif, 2022-05-09:
ok @org07 better installer for not get bugs... uninstall old  and install this [dtt4all_win_installer_2.0.1](https://github.com/Furtif/ddt4all/releases/tag/V2.0.1)

### `This pull request contains all dependencies updated with the latest versions:`
### `Dependencies :`
* [Python 3.9](https://www.python.org/downloads/release/python-390/) 
* [PyQt 5](https://pypi.org/project/PyQt5/)
* [pyusb](https://pypi.org/project/pyusb/)
* [pyserial](https://pypi.org/project/pyserial/)
* [crcmod](https://pypi.org/project/crcmod/)
* _**Hardware:**_ _An ELM327, ELS327 or OBDLink SX_ 
* [Python virtual environment](https://gist.github.com/dreamorosi/e2947827e5de92b69df68c88475eba38)

## `Cloning Source Code`
Next, you need to get the source code.  This source code repository uses git submodules. So when you clone the source code, you will need to clone recursively:

```
git clone --recursive https://github.com/cedricp/ddt4all.git
```

Or if you already cloned without the recursive option, you can update the submodules by running:

```
git clone --recursive https://github.com/cedricp/ddt4all.git
cd ddt4all
git submodule update --init --recursive
```

In cas for clone my branch is:
```
git clone --recursive https://github.com/Furtif/ddt4all.git
```
Or
```
git clone --recursive  https://github.com/Furtif/ddt4all.git
cd ddt4all
git submodule update --init --recursive
```

org07, 2022-05-09:
> tamam@org07hata almamak için daha iyi yükleyici... eskiyi kaldırın ve bunu yükleyin [dtt4all_win_installer_2.0.1](https://github.com/Furtif/ddt4all/releases/tag/V2.0.1)
> 
> ### `This pull request contains all dependencies updated with the latest versions:`
> ### `Dependencies :`
> * [Python 3.9](https://www.python.org/downloads/release/python-390/)
> * [PyQt 5](https://pypi.org/project/PyQt5/)
> * [pyusb](https://pypi.org/project/pyusb/)
> * [pyserial](https://pypi.org/project/pyserial/)
> * [crcmod](https://pypi.org/project/crcmod/)
> * _**Donanım:**_ _ Bir ELM327, ELS327 veya OBDLink SX_
> * [Python sanal ortamı](https://gist.github.com/dreamorosi/e2947827e5de92b69df68c88475eba38)
> 
> ## `Cloning Source Code`
> Ardından, kaynak kodunu almanız gerekir. Bu kaynak kod deposu, git alt modüllerini kullanır. Bu nedenle, kaynak kodunu klonladığınızda, yinelemeli olarak klonlamanız gerekecektir:
> 
> ```
> git clone --recursive https://github.com/cedricp/ddt4all.git
> ```
> 
> Veya özyinelemeli seçenek olmadan zaten klonladıysanız, aşağıdakileri çalıştırarak alt modülleri güncelleyebilirsiniz:
> 
> ```
> git clone --recursive https://github.com/cedricp/ddt4all.git
> cd ddt4all
> git submodule update --init --recursive
> ```
> 
> Klon için cas'ta şubem:
> 
> ```
> git clone --recursive https://github.com/Furtif/ddt4all.git
> ```
> 
> Veya
> 
> ```
> git clone --recursive  https://github.com/Furtif/ddt4all.git
> cd ddt4all
> git submodule update --init --recursive
> ```

you're super it's better like this

Furtif, 2022-05-09:
fine .

org07, 2022-05-09:
I installed ddt4all-win-installer_2.0.1, it closes without opening :(

Furtif, 2022-05-09:
> I installed ddt4all-win-installer_2.0.1, it closes without opening :(

have you istalled a DataBase ??

org07, 2022-05-09:
I don't know how to set up the database

Furtif, 2022-05-09:
[image]

Furtif, 2022-05-09:
is this you err ?

org07, 2022-05-09:
does not give an error message. shuts down before

org07, 2022-05-09:
> * c:\programfiles'e gidin (x86\ dtt4all klasörünü yeniden yükleyin
> * veritabanınızı c:\programfiles (x86\dtt4all) klasörüne kopyalayın
> * tüm bunlarla ilgili çelişkileriniz var burada iyi !!!
> * veritabanınız yoksa apk dosyasının ecus.zip dosyasını kullanın, bunu dtt4all klasörüne kopyalayın/yapıştırın.

ok

org07, 2022-05-09:
first I installed the old version, then I did not open it, I downloaded the ddt4all-win-installer_2.0.1 that you sent, I opened it in demo mode, no problem, I will try it on cars tomorrow. Sorry for making you tired, I'll let you know if it works tomorrow

Furtif, 2022-05-09:
her şeyi yanlış yaptın .... eski sürümü hiç kurmamalısın... yenisini kullanmak için bunu kaldırmalısın, sadece yeni kurulumla kullanmak için veritabanını yedeklemelisin 
Tanrı aşkına... her sene değiştirirsen, yine de senin sorunun bizde olacak...

org07, 2022-05-09:
I will remove the old version and install the new one, now

org07, 2022-05-09:
I uninstalled the old version, I installed the new version you gave me, I didn't make any copies, it works in demo mode, if it works in bluetooth, it's ok

Furtif, 2022-05-09:
normally it works with everything since I use dependency updates, there will be a problem if you go to the control panel and have a driver that is not ok this will no longer be a dtt4all problem but a windows driver problem if you have a driver with /!\ in the panel control devices see its ID and search in google in general there will be good things... other than that if it works with pyren the driver is ok only you have a mess that even you don't understand.

org07, 2022-05-09:
Another windows 10. I have a computer that I use for 32 bit CAN Clip, I tried it on it, it has the same problem. It works with pyren, but it does not work with ddt4all. Wired works on both my computers, but not on both computers with bluetooth. 32 and 64 bit

Furtif, 2022-05-09:
as I said above https://github.com/cedricp/ddt4all/issues/680#issuecomment-1121362648 this has the latest versions of dependencies, test it later if it doesn't work ok we will test some fixes that are in open pull request but you don't know how to do a git --clone or git --pull then I'll have to make your life easier with intallers..

Test before...

Furtif, 2022-05-09:
pyserial == manage COM ports ...
etc...

Furtif, 2022-05-09:
[image]

org07, 2022-05-10:
Computer Ddt4all does not run this device. This device runs PyReni. It runs both on an android phone. Can someone who knows the solution to this help?
this is the link i bought. https://tr.aliexpress.com/item/32572206764.html?spm=a2g0o.order_list.0.0.37cf3d12I7QB6o&gatewayAdapt=glo2tu
[image]
[image]
r

org07, 2022-05-10:
> normally it works with everything since I use dependency updates, there will be a problem if you go to the control panel and have a driver that is not ok this will no longer be a dtt4all problem but [...]

Thanks for your efforts, it doesn't work, also ddt4all-win-installer_2.0.1 is running very slow.

 Even the Renault CAN CLIP is faster.

I don't use the CLIP device as it is too slow, I just use it for Renault (Token) to auto promote the car via France
in mandatory times

AndyIreland, 2022-05-10:
I have that Kingbolen bluetooth make. And it works on my win 10 lenovo laptop  with DDT4ALL and I have split it open once and it does have the required PIC chip in it too

org07, 2022-05-10:
> 

As of now, I'm dealing with it. PyRen is working on all computers, but ddt4all is not working.

org07, 2022-05-10:
Someone must know this problem

AndyIreland, 2022-05-10:
you are doing it in the correct order? 

[image]

org07, 2022-05-10:
> you are doing it in the correct order?
> 
> [image]
[image]

Furtif, 2022-05-10:
so the boy is still having problems ??... as @ shows well in his image bluethooh is using COMx that is pyserial , now I don't know what your hardware is, many are ok with the version I gave above even mine is ok works perfectly I'll buy it here https://www.elm327.fr/
from what I see he goes to COM 4 instead of 3 .... COM4 is what? your telephone ? disable blue on your phone avoid interference

Furtif, 2022-05-10:
need maybe add something better like 'Use Item selected' in list of COM's

AndyIreland, 2022-05-10:
so in number 6 - no elm on com4 found . - does your pyREN work on com4? - if so do you exit totally out of PyREN before starting up DDT4ALL @org07  ?

AndyIreland, 2022-05-10:
if you download anydesk.exe from anydesk.com onto your windows laptop (with internet in your car area)  @org07 would you like me to remotely see if i can see anything stopping it working in DDT4ALL - if so give me a shout

Furtif, 2022-05-10:
or go inside your devices manager look for your Elm327 bluethooh properties -> port settings-> switch to COM 4

org07, 2022-05-10:
[image]

Furtif, 2022-05-10:
look @org07 @AndyIreland 
normally i not have rights to send this, but download full usual stuff, and says if ok i go delete this post  https://drive.google.com/file/d/1WrI39WV5jEtF4NuoF1kj-XGUjiyilW7V/view?usp=sharing

[52 more replies]

closed 2022-05-10 by org07