cedricp/ddt4all #686: Carminat Tomtom - No radio info bar on GPS screen

luissilva1971, 2022-05-26:
Hello, everyone.

I would like to ask your help i order to solve a problem on my megane 3 (ph1) sw 2010. Also a have a SD card (non oficial / prepared for new free maps). 
I have a non-bluetooth radio and GPS Carminat tomtom.  Also i have DDT4all program, and PyREN and a ELM327 with switch to change CAN1 /CAN2 reading.
Well, my GPS now does not show radio info bar and temp in the bottom as it was a few days ago. also if i change radio station or change radio volume, no info appears on screen. Radio plays as always.

"standard" GPS image
[image]

Atual GPS (no radio info bar...)
[image]

I have scanned the navigation ECU - NFA v0.6 - and  i have compared the values with a previos dump i made on PyREN on mod_ddt. And all the parameters are the same.

Previous parameters (from PyREN NFA dump):
[image]

scanned parameters yesterday (ddt4all)
[image]

Also radio has the same previous parameters.
Anyone knows how can this be solved?

Thank you in advance.

BarektE, 2022-05-28:
https://youtu.be/3HKBovSiJt8

luissilva1971, 2022-06-02:
Problem solved. Thank you @BarektE

closed 2022-06-02 by luissilva1971