cedricp/ddt4all #688: Error testing ELM

b0n0is, 2022-05-29:
Hello, 
I'm trying to activate Android auto on my Renault Clio 4 2017
I installed ddt4all with the database needed, I bough the ELM327 cable from Amazon [this one](https://www.amazon.fr/gp/product/B07NSPLVYW/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) 
and when I test the ELM i got some Errors, and I don't have Data when i connected to the car 
here is the errors, if some can help 
 
v1.0 ATZ [OK]
v1.0 ATE1 [OK]
v1.0 ATL0 [OK]
v1.0 ATI [OK]
v1.0 AT@1 [OK]
v1.0 ATAL [OK]
v1.0 ATBD [OK]
v1.0 ATBI [OK]
v1.0 ATCAF0 [OK]
v1.0 ATCFC1 [OK]
v1.0 ATCP 01 [OK]
v1.0 ATCS [OK]
v1.0 ATCV 1250 [OK]
v1.0 ATD [OK]
v1.0 ATDP [OK]
v1.0 ATDPN [OK]
v1.0 ATH0 [OK]
v1.0 ATI [OK]
v1.0 ATIB 10 [OK]
v1.0 ATIB 96 [OK]
v1.0 ATL1 [OK]
v1.0 ATM0 [OK]
v1.0 ATMA [ECHEC]
v1.0 ATMR 01 [ECHEC]
v1.0 ATMT 01 [ECHEC]
v1.0 ATNL [OK]
v1.0 ATPC [OK]
v1.0 ATR1 [OK]
v1.0 ATRV [OK]
v1.0 ATSH 001122 [OK]
v1.0 ATSH 012 [OK]
v1.0 ATSP A6 [OK]
v1.0 ATSP 6 [OK]
v1.0 ATCM 123 [OK]
v1.0 ATCM 12345678 [OK]
v1.0 ATCF 123 [OK]
v1.0 ATCF 12345678 [OK]
v1.0 ATST FF [OK]
v1.0 ATSW 96 [OK]
v1.0 ATSW 34 [OK]
v1.0 ATTP A6 [OK]
v1.0 ATTP 6 [OK]
v1.0 ATWM 817AF13E [OK]
v1.0 ATWS [OK]
v1.1 ATFC SD 300000 [OK]
v1.1 ATFC SH 012 [OK]
v1.1 ATFC SH 00112233 [OK]
v1.1 ATFC SM 1 [OK]
v1.1 ATPP FF OFF [OK]
v1.1 ATPP FF ON [OK]
v1.1 ATPPS [ECHEC]
v1.2 ATAR [OK]
v1.2 ATAT1 [OK]
v1.2 ATDM1 [ECHEC]
v1.2 ATIFR H [OK]
v1.2 ATIFR0 [OK]
v1.2 ATIIA 01 [OK]
v1.2 ATKW0 [OK]
v1.2 ATMP 0123 [ECHEC]
v1.2 ATMP 0123 4 [ECHEC]
v1.2 ATMP 012345 [ECHEC]
v1.2 ATMP 012345 6 [ECHEC]
v1.2 ATSR 01 [OK]
v1.3 ATCRA 012 [OK]
v1.3 ATCRA 01234567 [OK]
v1.3 ATD0 [OK]
v1.3 ATFE [OK]
v1.3 ATJE [OK]
v1.3 ATJS [OK]
v1.3 ATKW [OK]
v1.3 ATRA 01 [OK]
v1.3 ATS1 [OK]
v1.3 ATSP 00 [OK]
v1.3 ATV0 [OK]
v1.4 ATCEA [OK]
v1.4 ATCEA 01 [OK]
v1.4 ATCV 0000 [OK]
v1.4 ATIB 48 [OK]
v1.4 ATIGN [OK]
v1.4 ATPB 01 23 [OK]
v1.4 ATRD [OK]
v1.4 ATSD 01 [OK]
v1.4 ATSI [ECHEC]
v1.4 ATFI [ECHEC]
v1.4 ATSS [ECHEC]
v1.4 ATTA 12 [OK]
v1.4 ATCSM1 [OK]
v1.4 ATJHF1 [OK]
v1.4 ATJTM1 [OK]
v1.4 ATJTM5 [OK]
v1.4b ATCRA [OK]
v2.0 ATAMC [ECHEC]
v2.0 ATAMT 20 [ECHEC]
v2.1 ATCTM1 [ECHEC]
v2.1 ATCTM5 [ECHEC]
Adaptateur ARM incompatible 
Résultat: 79 réussis sur 95
Version ELM maximum:v1.4b

Thank you in Advance
B0n0is

AndyIreland, 2022-05-29:
you have to wire up a special cable or short out some pins in your ELM327 to read/write the rlink through the OBD2 socket

b0n0is, 2022-05-29:
> you have to wire up a special cable or short out some pins in your ELM327 to read/write the rlink through the OBD2 socket

thanks you for you answer, i did it following this schema [http://www.gps-carminat.com/v3/wp-content/uploads/2017/12/rallonge.png](url)
I bough this [cable](https://www.amazon.fr/gp/product/B073Y519FG/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) to do it easly 

I don't know if the problem caused by theses errors while testing the ELM

KarelSvo, 2022-05-29:
The ATPPS line shows that your Elm is not suitable for your application.

b0n0is, 2022-05-29:
if I understood correctly, my ELM cable is not the good one ? 
I have to take another ?

AndyIreland, 2022-05-29:
have you tried your adaptor with PyREN by any chance and ticking the 'cheap' chinese settings and see if it works with your USB ELM327? 
[image]

b0n0is, 2022-05-29:
I just installed PyRen to test my ELM327 (I don't have the databases)
I check ELM and I have the same errors as ddt4all 

Opening ELM
Checking port speed: 38400
Start COM speed:  38400
Changing baud rate to: 230400 ERROR - something went wrong. Let's back.
v1.0   ATZ                    [FAIL]
v1.0   ATE1                     [OK]
v1.0   ATL0                     [OK]
v1.0   ATI                      [OK]
v1.0   AT@1                     [OK]
v1.0   ATAL                     [OK]
v1.0   ATBD                     [OK]
v1.0   ATBI                     [OK]
v1.0   ATCAF0                   [OK]
v1.0   ATCFC1                   [OK]
v1.0   ATCP 01                  [OK]
v1.0   ATCS                     [OK]
v1.0   ATCV 1250                [OK]
v1.0   ATD                      [OK]
v1.0   ATDP                     [OK]
v1.0   ATDPN                    [OK]
v1.0   ATH0                     [OK]
v1.0   ATI                      [OK]
v1.0   ATIB 10                  [OK]
v1.0   ATIB 96                  [OK]
v1.0   ATL1                     [OK]
v1.0   ATM0                     [OK]
v1.0   ATCM 00000000            [OK]
v1.0   ATCF 00000000            [OK]
v1.0   ATCM 000                 [OK]
v1.0   ATCF 000                 [OK]
v1.0   ATMA                   [FAIL]
v1.0   ATMR 01                [FAIL]
v1.0   ATMT 01                [FAIL]
v1.0   ATNL                     [OK]
v1.0   ATPC                     [OK]
v1.0   ATR1                     [OK]
v1.0   ATRV                     [OK]
v1.0   ATSH 001122              [OK]
v1.0   ATSH 012                 [OK]
v1.0   ATSP A6                  [OK]
v1.0   ATSP 6                   [OK]
v1.0   ATST FF                  [OK]
v1.0   ATSW 96                  [OK]
v1.0   ATSW 34                  [OK]
v1.0   ATTP A6                  [OK]
v1.0   ATTP 6                   [OK]
v1.0   ATWM 817AF13E            [OK]
v1.0   ATWS                     [OK]
v1.1   ATFC SD 300000           [OK]
v1.1   ATFC SH 012              [OK]
v1.1   ATFC SH 00112233         [OK]
v1.1   ATFC SM 1                [OK]
v1.1   ATPP FF OFF              [OK]
v1.1   ATPP FF ON               [OK]
v1.1   ATPPS                  [FAIL]
v1.2   ATAR                     [OK]
v1.2   ATAT1                    [OK]
v1.2   ATDM1                  [FAIL]
v1.2   ATIFR H                  [OK]
v1.2   ATIFR0                   [OK]
v1.2   ATIIA 01                 [OK]
v1.2   ATKW0                    [OK]
v1.2   ATMP 0123              [FAIL]
v1.2   ATMP 0123 4            [FAIL]
v1.2   ATMP 012345            [FAIL]
v1.2   ATMP 012345 6          [FAIL]
v1.2   ATSR 01                  [OK]
v1.3   ATCRA 012                [OK]
v1.3   ATCRA 01234567           [OK]
v1.3   ATD0                     [OK]
v1.3   ATFE                     [OK]
v1.3   ATJE                     [OK]
v1.3   ATJS                     [OK]
v1.3   ATKW                     [OK]
v1.3   ATRA 01                  [OK]
v1.3   ATS1                     [OK]
v1.3   ATSP 00                  [OK]
v1.3   ATV0                     [OK]
v1.4   ATCEA                    [OK]
v1.4   ATCEA 01                 [OK]
v1.4   ATCV 0000                [OK]
v1.4   ATIB 48                  [OK]
v1.4   ATIGN                    [OK]
v1.4   ATPB 01 23               [OK]
v1.4   ATRD                     [OK]
v1.4   ATSD 01                  [OK]
v1.4   ATSI                   [FAIL]
v1.4   ATFI                   [FAIL]
v1.4   ATSS                   [FAIL]
v1.4   ATTA 12                  [OK]
v1.4   ATCSM1                   [OK]
v1.4   ATJHF1                   [OK]
v1.4   ATJTM1                   [OK]
v1.4   ATJTM5                   [OK]
v1.4b  ATCRA                    [OK]
v2.0   ATAMC                  [FAIL]
v2.0   ATAMT 20               [FAIL]
v2.1   ATCTM1                 [FAIL]
v2.1   ATCTM5                 [FAIL]
****************************************
*       RESETTING ELM
****************************************
*
*       ERRORS STATISTIC
*
* error_frame      =  0
* error_bufferfull =  0
* error_question   =  22
* error_nodata     =  0
* error_timeout    =  0
* error_rx         =  0
* error_can        =  0
*
*       RESPONSE TIME (Average)
*
* response_time    =  0.034
*
****************************************

Uncompatible adapter on ARM core
 pyren would not work with it

Result: 78 from 95
 Max version:v1.4b

I'm reading that there is some copy of the ELM327 cable that I have. i will try to order another to test, I'm thinking about a wifi one

AndyIreland, 2022-05-29:
I am using a bluetooth Kingbolen with required PIC chip in it:

https://www.amazon.co.uk/Kingbolen-Bluetooth-Scanner-Diagnostic-Compatible/dp/B07LC8LKRP

[image]
[image]

But regardless of the ELM test did you still experiment and try pyren with the boxes checked and un-checked to see if it would work anyway with your USB ELM327?

b0n0is, 2022-05-30:
I will try to find the PyRen Database needed and test that this evening

AndyIreland, 2022-05-30:
if you download pyren 184 from here it will have the ecus folder included and place a shortcut on your desktop :
https://www.drive2.com/b/501493851572142178/

b0n0is, 2022-05-30:
I just tested with PyRen and doesn't work nothing detected 

[image]

I will send back the ELM cable and buy another one 
I'm thinking about [this one](https://www.amazon.fr/Forscan-Scanner-ELMconfig-Diagnostic-commutateur/dp/B07MQ8GHG3/ref=sr_1_16?keywords=elm327+obd2&qid=1653940070&s=automotive&sprefix=ELM%2Cautomotive%2C90&sr=1-16)
or [this one](https://www.amazon.fr/FORScan-ELMconfig-Adaptateur-Scanner-diagnostic/dp/B08GPRWM5C/?_encoding=UTF8&pd_rd_w=NPq81&content-id=amzn1.sym.c5f1cc28-9183-45fd-860f-c83188e695b0&pf_rd_p=c5f1cc28-9183-45fd-860f-c83188e695b0&pf_rd_r=SE29EK44XQMRZMT2MEAJ&pd_rd_wg=1k1DX&pd_rd_r=b92b20d9-cc72-4d72-8bcb-29623a3e3d9c&ref_=pd_gw_ci_mcx_mr_hp_atf_m)

KarelSvo, 2022-05-30:
https://www.amazon.fr/ScanTool-425801-obdlink-SX-USB-professionnel/dp/B005ZWM0R4

b0n0is, 2022-05-30:
> https://www.amazon.fr/ScanTool-425801-obdlink-SX-USB-professionnel/dp/B005ZWM0R4

thank you so much for your help 
I ordred it, I should receive it tomorrow. i will test tomorrow evening

AndyIreland, 2022-05-30:
> I just tested with PyRen and doesn't work nothing detected
> 
> [image]
> 
Oh sorry about that I thought the ECU database came with pyren184 from drive2 site - must look again - or maybe it is the cable not detecting the ecu's in your car (although it does say scanning 14 out of 90) so it looked like it scanned /detected 14 ECU's in your car ... very strange.

AndyIreland, 2022-05-30:
> https://www.amazon.fr/ScanTool-425801-obdlink-SX-USB-professionnel/dp/B005ZWM0R4

Thats a dear one that - it does not even say it has the correct PIC chip in it , mind you in the reviews someone has put it works perfectly with DDT4ALL so thats a positive thing anyway.

KarelSvo, 2022-05-31:
> > I just tested with PyRen and doesn't work nothing detected

I don't know what you did, but you can see it in the log files.
When a selection of vehicle types is displayed, a database is also built in.

b0n0is, 2022-05-31:
Cable received, and work directly without installing any driver one Windows 10
connection one the car with Y cable PIN changed, and all it ok, reading information, and writing 
Android auto is now enable in my Clio 4 
Thank you all for your help @AndyIreland @KarelSvo

AndyIreland, 2022-05-31:
excellent!

closed 2022-06-01 by b0n0is