cedricp/ddt4all #691: Rlink1 2012 MID Radio detection

garzazum, 2022-06-11:
Hi, It's regarding Laguna III Rlink (Bose) and MID "Radio 2012". DDT4all when scanning for "x91 - Laguna III" does not recognize MID radio (it's not even on Audio list, seems not to be simply updated), when choosing "x43 - Laguna III (tricorps)" (radio is natively on the list in Audio module. but still in both cases scanning stops on 80% and not go further (it's not looking for others modules like BIC (it can recognize it only if Radio is not physically connected.  Below scanning error shows in output...
[image]

Problem is that I've replaced radio (MID to MID, old physically died) and on the welcome screen getting "unconfigured radio" VIN/ CRC updated (same as other values), but somehow afraid that fact that radio can not be correctly recognized, new values are "recorded" incorrectly so RLINK somehow can not correctly recognize new radio and keep saying that it's unconfigured.... any tips in this area please ? (already tried with pyren with ddt - but no positive results yet.

KarelSvo, 2022-06-12:
Submit your log files.
Are you using an adapter for Can Pin 12 and 13? https://www.drive2.ru/l/468057702971474426/

garzazum, 2022-06-12:
Hi, yes multimedia adapter is there. Navigation MFD is discovered and possible to modify (Android auto enabled), same BIC; Nav discovered on red, whilst BIC...v24 on green. Gathered everything (wiped vin and crc); in logs you should see language changes for audio and MFD; It's Laguna Ph3 2015 in native applications ddt2000 and Renolink (with same base 10.2021) "Radio 2012...." xml's are present with this model; here in latest DDt4all need to go to "tricopr" to see right radios; But still even if choose "tricorp" Audio error appears during module detecting.  
What is more... on the radio top there is extra label says "New Drive CDMM 10.5" seen such radios already, but could be that some newer firmware is present there. And this could be main reason ddt can not recognize it during scan and "unconfigured radio" 
[ecu_ddt.txt](https://github.com/cedricp/ddt4all/files/8886523/ecu_ddt.txt)
[elm_ddt.txt](https://github.com/cedricp/ddt4all/files/8886524/elm_ddt.txt)
[screens.txt](https://github.com/cedricp/ddt4all/files/8886525/screens.txt)
[x91.pdf](https://github.com/cedricp/ddt4all/files/8886527/x91.pdf)

KarelSvo, 2022-06-12:
Your MFD responds to the identification request,  correctly with the identification data.
inquiry  2180   answer  61 80 35 31 30 34 52 14 33 32 39 30 30 30 30 30 2D 3C 41 31 00 00 02 00 00 88 FF 

Your MID sends a valid response, but this does not contain any identification data.
inquiry  2180   answer 61 80 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 81
So you cannot identify the MID file with scanning.
Can you manually link the MID to the Radio 2012 DAB v4.0 file?
Then show Screen 26 - Traceability.

garzazum, 2022-06-13:
Hi,
In fact there is ffff under "software number $2180", same in attached bump from pyren-ddt
It this "software no" value is something that Rlink reads and require in similar way as ddt to recognize radio, it could be reason why it still keep saying "unconfigured radio" ?
[image]
[image]
[1655102806_Radio_2012_V2.4_20130604T134716.txt](https://github.com/cedricp/ddt4all/files/8888851/1655102806_Radio_2012_V2.4_20130604T134716.txt)

KarelSvo, 2022-06-13:
Was the MID made virgin by a configuration reset?
Or the MID has never been installed in a vehicle and the automatic configuration must first be carried out online with Renault Clip?

garzazum, 2022-06-13:
From the beginning it was "unconfigured radio" there was no VIN/CRC on it at all (radio is from different car - Latitude), putting vin/crc and other values has not change anything, then made "configuration reset" from ddt (radio) module thinking that putting everything from scratch could change something - no results. Initially when connected to car/rlink all buttons ware on lights on, after putting configuration via ddt, it was off and light on only for few sec during start-up, after "configuration reset" same story all buttons light ON.... putting configuration lights off... etc....  Most possible it was clear out before selling (no vin/crc other values) 
Is it like the only way is reprog via clip/ pushing new firmware ? 
Is it somehow possible to modify dump adding "software number" from different similar dump and try to rolback in pyren/ddt ?
br

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif