cedricp/ddt4all #694: Audio ECU skipped

rmeletti, 2022-06-23:
Hi guys, I'm trying to change screen orientation on my Megane IV. The screen is already installed but when running the ECU Scan the following problem appears:

Scanning ECU Audio
Skipping CAN extended address (not supported yet)  15

The radio is a replacement to the faulty original one which had the chassis verification cloned by a technical service provider.

Below is the full log in case somene can give me a clue on what is happening and what has to be done.

Cheers!
[Log.txt](https://github.com/cedricp/ddt4all/files/8967854/Log.txt)

AndyIreland, 2022-06-24:
do you have the specially wired to read CAN2 through the OBD2 socket and what DDT database version are you running with DDT4all maybe that needs updating?

rmeletti, 2022-06-24:
Yes! I just realized something like it yesterday and I'm already working on
a rig to switch OBD2 pins to CAN2. It is curious though that most tutorials
on internet assume the newbie knows this. I fount it in a German forum
reference and interpreted it might be the case.

Let's see how it goes! Thanks for the help!

rmeletti, 2022-06-25:
Done!

[image]

This €22 OBD did the trick:

[image]

Thank you!

rmeletti, 2022-06-25:
Nice software, thank you!

closed 2022-06-25 by rmeletti