cedricp/ddt4all #695: Obdlink SX COM port Problem

EricTCartman, 2022-07-01:
hello guys and thanks for the hard work :)

i have a problem with ddt4all and obdlink sx cable it does not connect to the car (laguna 3)
i want to mention that renolink app works just fine without problems but not for ddt4all. 
sometimes it would give me permission error could not open port 'COM4': PermissionError(13, 'Access is denied.', None, 5)
and other times it would do this
Initilizing ELM with speed 115200...
Trying to open portCOM4 @ 115200
Checking port speed: 500000
ELM not responding
Port closed
Trying to open portCOM4 @ 38400
Checking port speed: 500000
ELM not responding
Port closed
Trying to open portCOM4 @ 115200
Checking port speed: 500000
ELM not responding
Port closed
Trying to open portCOM4 @ 230400
Checking port speed: 500000
ELM not responding
Port closed
Trying to open portCOM4 @ 57600
Checking port speed: 500000
ELM not responding
Port closed
Trying to open portCOM4 @ 9600
Checking port speed: 500000
ELM not responding
Port closed
Trying to open portCOM4 @ 500000
Checking port speed: 500000
ELM not responding
Port closed

i tried it connect to the car too and same responses, all drivers installed FTDI. any help is welcomed. Thanks so much!

KarelSvo, 2022-07-03:
Give me your used elm.py file to change.

benderfuturama, 2022-12-11:
You need to change baudrate to 115200, renolink is using 1000000.

KarelSvo, 2022-12-12:
Simply add the missing baud rates in elm.py ;-)

benderfuturama, 2022-12-14:
Will all functions of ddt4all work? Maybe they require 115200?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

dido17diag, 2024-05-31:
have a problem with ddt4all and obdlink sx cable it does not connect

Duke-Jones, 2024-09-04:
> have a problem with ddt4all and obdlink sx cable it does not connect

Try to do a factory reset: 
[https://support.obdlink.com/support/solutions/articles/43000721545-restore-obdlink-wired-adapter-to-factory-settings](https://support.obdlink.com/support/solutions/articles/43000721545-restore-obdlink-wired-adapter-to-factory-settings)

It has helped me